makerbase-mks/MKS-DLC32 #1: Homing does not work on 2 axis machine with hard limits

Poppy45, 2021-09-06:
homing does not work on 2 axis machine with hard limits .so need to modify firmware to stop x axis homing as per grbl instructions'.

mks-viva, 2021-09-07:
Please provide a video when your manchie homing.

Poppy45, 2021-09-07:
https://user-images.githubusercontent.com/64296414/132287709-65737b8e-4093-4b42-bb63-c7bf4416c18f.mp4

https://user-images.githubusercontent.com/64296414/132287731-7ec0ba69-0673-497d-9a13-53f34aea8c7f.mp4

goes into alarm 8 as tries to home z axis but does not exist

Poppy45, 2021-09-07:
closed by mistake

Poppy45, 2021-09-07:
as per link https://github.com/gnea/grbl/wiki/Two-Axis-System-Considerations

solawc, 2021-09-07:
> as per link https://github.com/gnea/grbl/wiki/Two-Axis-System-Considerations

Hello, although dlc32 supports the instructions and upper computer of grbl, some operations have been changed. For dlc32 just delivered, Z-axis and z-limit are not used in the firmware, so they will not start the effect. Please check whether the level setting of a lower limit bit is correct

Poppy45, 2021-09-07:
[DLC32 setting.txt](https://github.com/makerbase-mks/MKS-DLC32/files/7120574/DLC32.setting.txt)
Thanks that was i was asking. Please explain what you mean by  level setting of a lower limit bit is correct. attached is my setting in txt file

Poppy45, 2021-09-25:
got it working using NC switches need to have $5=0 not 1 which is reverse of grbl setting

Poppy45, 2021-09-25:
thanks for non help

closed 2021-09-25 by Poppy45

Poppy45, 2021-09-28:
read the solution above $5=0 with NC switches not as per grbl instructions which states $5=1