makerbase-mks/MKS-DLC32 #2: TFT35 V2 Rotation

Emrah8551, 2021-09-06:
How can i rotate the screen on the new MKS TFT3.5 V2.0?
Thanks

mks-viva, 2021-09-07:
@solawc Please evaluate it.

solawc, 2021-09-07:
> How can i rotate the screen on the new MKS TFT3.5 V2.0?
> Thanks

This setting is not available at present. It needs to be updated in a later version

MrErdnuss, 2023-03-07:
Is this still not possible in v2.3? With the Longer Ray5 there is no way to mount the display the other way around, so the only option is to rotate the display via software.

ciaky73, 2023-03-19:
Has anyone found the solution to rotate the display screen?

ciaky73, 2023-03-19:
> Is this still not possible in v2.3? With the Longer Ray5 there is no way to mount the display the other way around, so the only option is to rotate the display via software.

I also own the loger ray5 and wanted to implement autofoscus on z axis. Currently it is not possible to do so with the original firmware. With the makerbase firmware the z-axis works but the display is turned upside down.

andreasr81, 2023-07-14:
no solution so far? i have a ray5 as well.

KeithLongerLaser, 2023-08-09:
Why has this not been addressed yet?!!!!!

kris0725pl, 2024-03-07:
> I also own the loger ray5 and wanted to implement autofoscus on z axis. Currently it is not possible to do so with the original firmware. With the makerbase firmware the z-axis works but the display [...]

Hi...what version you loading to ray5?

KeithLongerLaser, 2024-03-12:
You can update this firmware and set it using $49

0 and 1, the screen will flip.

kris0725pl, 2024-03-12:
i dont have option $49 How to add? i have 45...50...etc

KeithLongerLaser, 2024-03-13:
You need to update your firmware using the file I attached

KeithLongerLaser, 2024-03-13:
> I have the same Problem, @KeithLongerLaser <https://github.com/KeithLongerLaser>  I can't see a attached Firmware?
>
> Regards

There you go 😊

kris0725pl, 2024-03-13:
Where is this attached file? You have eyes? Brain? Your post is nothing worst. Sory...waste my time.

KeithLongerLaser, 2024-03-14:
[DLC32_Laser_v2.2_2023081601.zip]( https://github.com/makerbase-mks/MKS-DLC32/files/14608792/DLC32_Laser_v2.2_2023081601.zip ) **Here is the firmware file in zip format**
[RAY5 Firmware Upgrading Guide 1.0.pdf]( https://github.com/makerbase-mks/MKS-DLC32/files/14608794/RAY5.Firmware.Upgrading.Guide.1.0.pdf ) **Guide to upgrading the firmware**

kris0725pl, 2024-03-16:
this firmware sucks...on longer ray 5. why? because screen is fleep. no available parameter $49 as you say in config .nc

KeithLongerLaser, 2024-03-17:
Go read the original thread! That was sent to me by Longer!

andreasr81, 2024-03-17:
i dident test it, but did you try $49 in the console, light burn or other software?
I mount my display 180° it works but is not so nice.
maybe I cad a new case, that fits a flipped screen

kris0725pl, 2024-03-17:
No...this firmware is not for longer ray5 20W. I tester IT. I fleep screen...looks fine. But I have error message...beeper always beep. I reset error but touch screen not response for any touch.

For me this is crap firmware. I upload original From longer.