makerbase-mks/MKS-DLC32 #5: Where is the schematic?

automics, 2021-09-10:
Why not post the schematic?
Lack of scheme strains me greatly to make the order.
I don't want to buy a cat in a bag.

mks-viva, 2021-09-11:
MKS will upload it

closed 2021-09-11 by mks-viva

makerbase-mks, 2021-09-11:
@automics You can find it here:
https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V1.1_001/MKS%20DLC32%20V1.1_001%20SCH.pdf

HAL9000-ROOT, 2025-06-16:
> [@automics](https://github.com/automics) You can find it here: https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V1.1_001/MKS%20DLC32%20V1.1_001%20SCH.pdf

dead link: 404 - page not found
The 

main
 branch of 

MKS-DLC32
 does not contain the path 

hardware/MKS DLC32 V1.1_001/MKS DLC32 V1.1_001 SCH.pdf.