makerbase-mks/MKS-DLC32 #6: Firmware upload not working

trashypopashy, 2021-09-12:
I can not upload the CNC_normal Firmware to the DLC32.
[image]

mks-viva, 2021-09-13:
@solawc Please check if his settings are correct?

Igoravg98, 2021-09-13:
[image]
I'm having a problem uploading the firmware too, the address is always red like this

solawc, 2021-09-13:
> [image]
> I'm having a problem uploading the firmware too, the address is always red like this

From your prompt, it seems that the serial port you selected is incorrect. Normally on the tool, if the selected serial port and address are correct, it should display green instead of red. Please check if the path of .bin is correct? Address should fill in 0x00 instead of 0 directly

Igoravg98, 2021-09-13:
[image]
Now I noticed that the button to add the bin file is not working, this appears here when I try to add the file

solawc, 2021-09-13:
> [image]
> Now I noticed that the button to add the bin file is not working, this appears here when I try to add the file

Since this tool is not developed by us, it is an official tool of ESP32, and I am not aware of its problems. But you can enter a complete path directly above, it is no problem.

Igoravg98, 2021-09-13:
It was a problem with my computer, I tried it on another one and everything worked out fine, I'm really happy, thanks for trying to help me!

solawc, 2021-09-13:
> It was a problem with my computer, I tried it on another one and everything worked out fine, I'm really happy, thanks for trying to help me!

Oh, it’s okay, I’m glad to help you solve the problem

trashypopashy, 2021-09-13:
I solved my problem: I needed not just to start the flashing tool with admin rights, but first change the files properties >> Security >> to "full control"

closed 2021-09-15 by makerbase-mks