makerbase-mks/MKS-DLC32 #7: Motor-Port always with 24V

trashypopashy, 2021-09-27:
No matter what I do: On the Motor-Port I always geht 24 V.

Is my Mosfet broken?

Would it be possible (with the CNC-Firmware), to get the Motor-Signal on the TTL or Laser-Port?

mks-viva, 2021-09-28:
Hi! I want to know, if you power on for MKS DLC32 board, the mosfet led is light up?

trashypopashy, 2021-09-28:
Yes, it does. I also reflashed the CNC-firmware, but the Problem stays the same.

mks-viva, 2021-09-28:
Can you provide a video about when power on.

trashypopashy, 2021-09-28:
#

trashypopashy, 2021-09-29:
I have an external pwm-to-analog (0-10V)-converter for my seperate spindle-driver, so the TTL-Signal would be okay for me. Is there a way to get the TTL working in the CNC-firmware?

mks-viva, 2021-09-29:
[image]

trashypopashy, 2021-09-29:
I know where the port is located, but it does not work. Maybe all my issues with the DLC32 are because of the CNC-firmware?

mks-viva, 2021-09-29:
Can you use laser firmware test board and it is will work?

trashypopashy, 2021-09-29:
I can give it a try. But I need the Z-axis, of course (because I have a CNC).

trashypopashy, 2021-09-29:
With the Laser-firmware on the TTL-Port I get the following output:
- M3 S0 ---> TTL=0%
- M3 S5 ---> TTL=95%
- M3 S50 ---> TTL=50%
So this is inverted.

With the CNC-Firmware I get the following:
- M3 S0 ---> TTL=0%
- M3 S5 ---> TTL=0%
- M3 S10 ---> 0%
- M3 S15 ---> 1,2%
- M3 S20 ---> 4,7%
- M3 S25 ---> 27%
(- M3 S26 ---> 40%
- M3 S27 ---> 61%
- M3 S28 ---> 80%
- M3 S29 ---> 95%)
- M3 S30 ---> 95,7%
- M3 S35 ---> 95,5%
- M3 S40 ---> 95%
- M3 S45 ---> 94,6%
- M3 S50 ---> 94,1%
- M3 S55 ---> 93,6%
- M3 S60 ---> 93,2%
- M3 S65 ---> 92,7%
- M3 S70 ---> 92,2%
- M3 S75 ---> 91,7%
- M3 S80 ---> 91,2%
- M3 S85 ---> 90,7%
- M3 S90 ---> 90,2%
- M3 S95 ---> 89,7%
- M3 S100 ---> 89,2%

trashypopashy, 2021-10-09:
trashypopashy@gmx.de
.

makerbase-mks, 2021-10-09:
@trashypopashy You can delete email post now

makerbase-mks, 2021-10-09:
[DLC32_CNC_V1.2.001_Normal_8M_H35_1009.bin.zip](https://github.com/makerbase-mks/MKS-DLC32/files/7315054/DLC32_CNC_V1.2.001_Normal_8M_H35_1009.bin.zip)
DLC32 firmware for CNC, pwm default is 1khz, and can be modified using $28, for example, 1kz: $28 1000

trashypopashy, 2021-10-10:
I need to reopen it.

Okay, so now, that it is at 1kHz...

My Spindle is spinning between the following commands:
- M3 S260.9 which equals 350 rpm (lower input is too low for the spindle which is okay)
- M3 S1100 which equals 12500 rpm which is the highest rpm of the spindle, so that is fine as well.

Now I want to know what needs to be changed, so that I can send the command M3 S5000 and the spindle would turn with 5000 rpm / M3 S350 and the spindle turns with 350 rpm / M3 S11800 and the spindle turns with 11800 rpm ...and so on...

My parameters set in the DLC_CFG.txt that belong to the spindle are:
- $30=12500.000 	
- $31=0.000 		
- $32=1

Mar4M, 2023-10-14:
@makerbase-mks Hi. Is this problem related to mine? My spindle motor is always turned on after power on and cant be controlled by UGS or Candle and I have no way of controlling the spindle while the power is on. Have been using the 1st version o the firmware for CNC since 2021 and directly updated to this version - V2.1.0_H35_20220526_02_N_CNC. Please help. By the way, I'm just a maker with no or at most very minimal knowledge with coding/programing.