makerbase-mks/MKS-DLC32 #8: WiFi unconnected after reboot

trashypopashy, 2021-09-27:
Is there a way to mak the DLC32 reconnect to the known WiFi-Network after a reboot or just on any power-up?

mks-viva, 2021-09-28:
@solawc 
Please evaluate this feature.

Plottmania, 2021-09-28:
Quite simply, after every restart or reset you have to re-enter the WiFi connection.
It's not fun and annoying.
Then there are translation errors or Chinese characters in the "English" language.

solawc, 2021-09-29:
> 有没有办法让 DLC32 在重新启动或任何上电后重新连接到已知的 WiFi 网络？

> Is there a way to mak the DLC32 reconnect to the known WiFi-Network after a reboot or just on any power-up?

Wi-Fi automatic connection is possible, but what I have to consider is that if the connection fails, or other conditions cause problems when connecting to Wi-Fi in the background. This will give a solution in the next update.

automics, 2021-10-03:
Please do not forget to update the firmware for the CNC ! (WiFi automatic connection for STA mode)

trashypopashy, 2021-10-04:
When will the firmware be updated?

zoldrax, 2021-10-06:
I planned to use the board for CNC without a display over Wi-Fi. But now I must send [ESP115]ON command over USB every reset. After reset or power failure, the value  [ESP115] goes to OFF again.

derFrickler, 2021-10-07:
My version is just Grbl_Esp32, nu Touch support, but changable spindle PWM via $Spindle/PWM/Frequency
https://github.com/bdring/Grbl_Esp32/wiki/Spindle-Types#spindletypepwm
Laser support see: https://github.com/bdring/Grbl_Esp32/wiki/Laser-Mode

Plottmania, 2021-10-07:
nicht wollen....ich hab einen und dazu noch reichlich DLC 2.0 und 2.1 mit diversen Treibern und Displays : )
Kann man sich anschauen unter www.plottmania.de (Deutschsprachig)

Regards Stephan

trashypopashy, 2021-10-09:
> > > > 有没有办法让 DLC32 在重新启动或任何上电后重新连接到已知的 WiFi 网络？
> > > 
> > > 
> > > > Is there a way to mak the DLC32 reconnect to the known WiFi-Network after a reboot or just on any power-up?
> > > 
> > > 
> > > Wi-Fi automatic connection is possible, but what I have to consider is that if the connection fails, or other conditions cause problems when connecting to Wi-Fi in the background. This will give a solution in the next update.
> > 
> > 
> > Is this already implemented in the CNC-firmware, You uploaded 2 hours ago?
> 
> The current firmware only fixes some bugs on the control or interface, and has not added the function of automatically connecting WiFi; The reason why the original version is not added is that it cannot be connected at one time, but needs to be connected many times. If the number of times is not good, the connection failure may be very high, but too many times may affect the operation or prolong the startup time; This feature will be added and is currently being tested

Maybe you could implement a possibility to choose automatic or manually (re-)connect in the LCD-menu.

killrob, 2021-10-18:
there is a way to connect wifi without using TFT, that i have not bougth?

trashypopashy, 2021-11-11:
@solawc You updated the firmware for Laser-machines, but not for CNC-machines. Could You please also upload a firmware-update in the CNC-folder?

Is the WiFi-functionality improved the way, we discussed in this thread?

alfrankgit, 2021-11-13:
With the current firmware DLC32_V2.0_D2.00.3_H35_8M_1108_C the WiFi reconnects automatically after power off / on.

alfrankgit, 2021-11-13:
> Is there a way to make the DLC32 reconnect to the known WiFi-Network after a reboot or just on any power-up?

With the current firmware DLC32_V2.0_D2.00.3_H35_8M_1108_C the WiFi reconnects automatically after power off / on.

trashypopashy, 2021-11-17:
> With the current firmware DLC32_V2.0_D2.00.3_H35_8M_1108_C the WiFi reconnects automatically after power off / on.

But that is only for lasers. @solawc  I need the CNC-firmware.

makerbase-mks, 2021-11-25:
@trashypopashy Has fixed at the latest firmware. Please test and reopen the issue if you still has problem. https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/CNC/Normal

closed 2021-11-25 by makerbase-mks

RandomHacks-Git, 2021-12-06:
@makerbase-mks what about the 2.4" laser version? It doesn't automatically connect still, could you fix it?