makerbase-mks/MKS-DLC32 #10: Not sourde code

JuananBluenoa, 2021-10-01:
Please, could you post the source code? 

Tanks

automics, 2021-10-02:
I need the source code too, because I do not use the external display, but I need a constantly enabled WiFi and telnet port.

m3dkata, 2021-10-05:
Yes, we need source code, count me.

derFrickler, 2021-10-08:
Same board revision for me. Try formatting the full chip once before flashing:
../linux-amd64/esptool erase_flash

makerbase-mks, 2021-10-09:
@derFrickler @automics @GadgetX @JuananBluenoa @m3dkata MKS DLC32 will be open souce recently, we are preparing it.

derFrickler, 2021-10-09:
I have a DLC32 v1.1 and TTL for Laser/Spindle is on IO22.
Says the chematixcs (labeled LC) https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V1.1_002/MKS%20DLC32%20V1.1_002%20SCH.pdf
As well as my measurements on the TTL output when i run "M03 S500" on my code.

GadgetX, 2021-10-09:
> I have a DLC32 v1.1 and TTL for Laser/Spindle is on IO22.
> 
> Says the schematics (labeled LC) https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V1.1_002/MKS%20DLC32%20V1.1_002%20SCH.pdf
> 
> As well as my measurements on the TTL output when i run "M03 S500" on my code.
> 
> io19 is LCD_MISO which i use for control_reset_pin

Yup I understand pin out on schematic it's different to pin out on the board definitely MKS give shit on this what we write !!!￼￼

diruuu, 2021-12-02:
@derFrickler do you have config example for FluidNC compatible for MKS-DLC32? Grbl_Esp32 is now become FluidNC but they're using different config file format.

trashypopashy, 2021-12-05:
> @derFrickler @automics @GadgetX @JuananBluenoa @m3dkata MKS DLC32 will be open souce recently, we are preparing it.

When will you do this? Come on, guys! Go ahead!

daxliniere, 2022-04-23:
@makerbase-mks , how is the source code coming along? We understand that your use of Grbl_ESP without releasing the source code is a breach of the Grbl_ESP licence agreement. This was confirmed by Bart Dring, author of Grbl_ESP.

makerbase-mks, 2022-11-15:
Code open source late but finally here: https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

closed 2023-01-09 by abinoer