makerbase-mks/MKS-DLC32 #12: How to turn on WIFI via cmd?

automics, 2021-10-03:
Hi, I have no touch display, how to enable WiFi through the console?
> $I
[VER:1.1.2021051202:]
[OPT:PHSW]
[MSG:Using machine:MKS DLC32]
[MSG:No Wifi]

JuananBluenoa, 2021-10-03:
[https://github.com/bdring/Grbl_Esp32/wiki/Settings](https://github.com/bdring/Grbl_Esp32/wiki/Settings)

derFrickler, 2021-10-03:
Are you sure?
Guess it's not running grbl_esp32

automics, 2021-10-03:
[ESP115] - Turns the WiFi ON or OFF

No connection, why is SSID not displayed?
$I
[VER:1.1.2021051202:]
[OPT:PHSW]
[MSG:Using machine:MKS DLC32]
[MSG:Mode=STA:SSID=:Status=Not connected:IP=0.0.0.0:MAC=7C-87-CE-12-9B-A8]
ok

JuananBluenoa, 2021-10-03:
to connect to an access point (STA mode)
$ESP100=YOUR_SSID     or     $Sta/SSID=YOUR_SSID
to create an access point (AP mode)
$ESP105=newssid     or     $Ap/SSID=newssid

automics, 2021-10-03:
After the reset, the connection was established!

[ESP444]RESTART
...
[ESP115]ON
[MSG:Client Started]
[MSG:Connecting winet]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connected with 172.16.0.41]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
ok
WebUI connected!

closed 2021-10-03 by automics

ahmetbozaci, 2021-11-12:
@automics Hello ı dnt have tft too. How did you open wifi. I check the this [page ](https://github.com/bdring/Grbl_Esp32/wiki/Settings) I try some codes but ı cant see board wifi in wifi list. I appreciate if you write how did you do that.

ahmetbozaci, 2021-11-12:
> @automics Hello ı dnt have tft too. How did you open wifi. I check the this [page ](https://github.com/bdring/Grbl_Esp32/wiki/Settings) I try some codes but ı cant see board wifi in wifi list. I appreciate if you write how did you do that. [image]

I connect finally.If somebody cant connect

```
1. $ESP444=RESTART
2. $ESP115=ON  
3. $ESP100=your_ssdi_name
4. $ESP101=your_ssdi_password
5. $ESP110=STA
6. $ESP444=RESTART 
7. $ESP115=ON
```

Maybe it can work after step 5 but for me ı have to restart. Then open wifi you will see ip address in grbl console.