makerbase-mks/MKS-DLC32 #17: syncronize spindle-speeds

trashypopashy, 2021-10-10:
Now, that it is at 1kHz...

My Spindle is spinning between the following commands:

M3 S260.9 which equals 350 rpm (lower input is too low for the spindle which is okay)
M3 S1100 which equals 12500 rpm which is the highest rpm of the spindle, so that is fine as well.
Now I want to know what needs to be changed, so that I can send the command M3 S5000 and the spindle would turn with 5000 rpm / M3 S350 and the spindle turns with 350 rpm / M3 S11800 and the spindle turns with 11800 rpm ...and so on...

My parameters set in the DLC_CFG.txt that belong to the spindle are:
$30=12500.000
$31=0.000
$32=1

solawc, 2021-10-14:
> Now, that it is at 1kHz...
>
> My Spindle is spinning between the following commands:
>
> M3 S260.9 which equals 350 rpm (lower input is too low for the spindle which is okay) M3 S1100 which equals 12500 [...]

Excuse me, do you use CNC? If yes, set $32 to 0

trashypopashy, 2021-10-15:
Sorry, but setting $32=0 did not change the behaviour.

solawc, 2021-11-25:
> @solawc or anybody else: Do you have an idea, how to configure the settings to match the M3 Sxxx-commands with the real speeds of the spindle?

Well, actually I don't quite understand what you mean. M3 SXXX actually controls the output of PWM, and XXX represents the percentage. What is the speed of your spindle type and the corresponding voltage? I don't know this.

kweilim98, 2022-04-05:
> Now, that it is at 1kHz...
>
> My Spindle is spinning between the following commands:
>
> M3 S260.9 which equals 350 rpm (lower input is too low for the spindle which is okay) M3 S1100 which equals 12500 [...]

Sorry sir, can u share how to connect your spindle to motherboard?

iDBaliMaker, 2022-06-06:
> Sorry sir, can u share how to connect your spindle to motherboard?

sir yes sir... me too have spindle problem speed synchronize..
wird problems..

trashypopashy, 2022-06-06:
I used a pwm to analog concerter: https://de.aliexpress.com/item/4001149885707.html

kweilim98, 2022-07-04:
> I used a pwm to analog concerter: https://de.aliexpress.com/item/4001149885707.html

Hi sir, have u even solve the problems with the pwm to analog converter that can synchronise the spindle speed as u settings? If yes can u share here how u solve it?

trashypopashy, 2022-07-04:
Sorry, no. Just by try an error. I wrote a table with g-Code rpm and real rpm by trying it out.

kweilim98, 2022-07-05:
> Sorry, no. Just by try an error. I wrote a table with g-Code rpm and real rpm by trying it out.

Ok, I see, thx u