makerbase-mks/MKS-DLC32 #18: no Z axis movement and unable to upload firmware

grumpyduck300, 2021-10-14:
controller will not move z axis on my machine and i keep getting a sync error when using the download tool. what can i do to fix this issue?

solawc, 2021-10-14:
> controller will not move z axis on my machine and i keep getting a sync error when using the download tool. what can i do to fix this issue?

Yes, for the current firmware, since the limit return mode is not configured, the return mode is x, y return to zero, and there is no Z, so the Z axis is shielded temporarily. This will be issued and the return configuration mode will be provided at the next function version update; If there is a problem with your download, please provide feedback error information so that I can evaluate the problem

derFrickler, 2021-10-14:
Try with a lower baud rate, hat problems flashing mine with 921600 as well.

grumpyduck300, 2021-10-14:
Tried the lower baudrate same issue if there any jumpers on the board that need to be adjusted to put the board into download mode?

killrob, 2021-10-18:
> 
> 
> I've had the same issue. I solved my problem like this: I needed not just to start the flashing tool with admin rights, but first change the files properties >> Security >> to "full control"

i still have full control on both files, the firmware and the .exe, and i ran it as admin but still same issue than @grumpyduck300 .
the only difference is that i run a newer version of the tool the 3.9.0

rlinta, 2021-10-20:
I have a similar problem. the firmware does not load - sync error.

ch0 - io0 (esp32 pin 25)
ch1 - reset (esp32 pin 3)
ch2 - probe/io2 (esp32 pin 24)

[image]

[image]

killrob, 2021-10-20:
> 
> 
> I have a similar problem. the firmware does not load - sync error.
> 
> ch0 - io0 (esp32 pin 25) ch1 - reset (esp32 pin 3) ch2 - probe/io2 (esp32 pin 24)
> 
> [image]
> 
> [image]

hi @rlinta , where do you found  the MKS ESP32 download tool?

Lazarus-Rising, 2021-11-12:
so is this the definitive answer then to the lack of movement on the Z axis?
I've uploaded the cnc firmware but still no Z axis movement.
can MKS confirm the connecting the capacitor to gnd etc is correct and will Z axis movement then be enabled in laser mode?

Anderson8153, 2021-11-16:
> este é o erro que recebo toda vez que tento fazer o upload do firmware cnc esperando que o z esteja habilitado
> 
> existe alguma maneira de fazer upload dos arquivos de configuração brutos para que possamos editar e compilar o firmware por nós mesmos?
> 
> # deslocamento de teste: 4096 0x1000
> caso ok
> # CONNECT BAUD: 115200
> ..... _****.....****_ ..... _****.....****_ ..... _****.....****_ ..... _****.....****_ ..... _****.....****_ [2021-10-14 06: 17: 58,473] [ESP8266Loader_spi [1]] [espDownloader.py] [linha: 351] [ERROR]: Erro de sincronização do chip: Falha ao conectar ao ESP32: Tempo limite excedido aguardando cabeçalho do pacote [2021 -10-14 06: 17: 58,474] [ESP8266Loader_spi [1]] [espDownloader.py] [linha: 439] [ERROR]: Erro de sincronização do chip ESP32 esp_sync_blocking.
> 
> [image]

Defina BAUD: 115200 e @=0
[image]

makerbase-mks, 2021-11-25:
@grumpyduck300 , the latest firmware has enabled the Z axis: https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/CNC/Normal. 

For upload problem, please refer to [#19](https://github.com/makerbase-mks/MKS-DLC32/issues/19#issuecomment-978785658)

makerbase-mks, 2021-11-25:
Please try, I close this issue now. Please reopen it if you still have problem.

closed 2021-11-25 by makerbase-mks