makerbase-mks/MKS-DLC32 #19: Not going into bootloader mode

grumpyduck300, 2021-10-15:
I have tried everything I can think of to get my board into bootloader mode and it wouldn't work. 
I can communicate with the board esptool.py runs but it stays in operational mode failing the sync. Any help will be appreciated. I can't use my laser tell I figure this out.

solawc, 2021-10-20:
> I have tried everything I can think of to get my board into bootloader mode and it wouldn't work. I can communicate with the board esptool.py runs but it stays in operational mode failing the sync. Any help will be appreciated. I can't use my laser tell I figure this out.

The download mode of ESP32 is controlled by the starting level of GPIO. If you ensure that the CH340 driver on the serial port and PC is complete, please help check the level of IO0, Probe at power-on, or they Is there an external device?

makerbase-mks, 2021-11-25:
@grumpyduck300 We found some computer's usb cannot support the valid signal to reset the board, even add the capacitor as @rlinta mentioned, some computers are still unavailable. If you cannot make it enter download mode, you can connect the two pins together(pull gpio0 to gnd), and power on the board to enter download mode. And disconnect the two pins and re-power the board to work normally.

[image]

closed 2021-11-25 by makerbase-mks

idilbaz, 2022-09-03:
> @grumpyduck300Bazı bilgisayarların usb'sinin kartı sıfırlamak için geçerli sinyali destekleyemediğini, hatta kapasitörü aşağıdaki gibi ekleyemediğini gördük.@rlintaBahsedilen, bazı bilgisayarlar hala kullanılamıyor. İndirme moduna giremezseniz, iki pimi birbirine bağlayabilir (gpio0'ı gnd'ye çekin) ve indirme moduna girmek için kartı açabilirsiniz. Ve iki pimin bağlantısını kesin ve kartın normal şekilde çalışması için yeniden güç verin.
> 
> [image]

There is no such place on my card. Where exactly is it ?

[image]