makerbase-mks/MKS-DLC32 #23: Connect to wifi without display

21mac, 2021-10-29:
Hi @all,
is there any chance that I can use the wifi feature without an LCD or TFT?
When I bought it I did not know that I need a display to use the features.

21mac, 2021-10-29:
Found the ESP-Commands and added my SSID via [ESP105] and my password via [ESP106] and after I turned on my radio [ESP115] i finally got an IP address. great!

I opened the android app "MKSLaser" and searched for the laser. the app found it on IP 192.168.137.140 (which is correct, [ESP107] gives me this IP].

BUT then i always get the error message "connect fail" in the app.

it does not matter if I change the  Radio/Mode [ESP110] to STA or AP.

Anyone got an idea what to do / check?

thanks

matejc88, 2021-11-07:
I have this problem too. And after i restar board, i shud again turn wifi on to connect board to wifi. It doesn connect automaticly.  And you can buy touchscreen only together with board....

ahmetbozaci, 2021-11-07:
I ask it to on mks youtube channel. He said software will update. I am waiting it too.

21mac, 2021-11-18:
Update: I got a newer version (MKS-DLC32 V2.0) and it worked.
I now have an LCD but even without it, it keeps the wifi settings.

closed 2021-11-18 by 21mac

ahmetbozaci, 2021-11-18:
> Update: I got a newer version (MKS-DLC32 V2.0) and it worked. I now have an LCD but even without it, it keeps the wifi settings.

Phone application can connect it? I cant connect. Only website is working.

21mac, 2021-11-18:
yes, the app works with the V2.0

but the old V1.1 did NOT work with the app.

ahmetbozaci, 2021-11-20:
> yes, the app works with the V2.0
> 
> but the old V1.1 did NOT work with the app.

Are you talking about the software version right? V2.0 board and v1.1 board is little different. Is it  possible to install software v2.0 on v1.1 board without any issue?

Kleitongie, 2021-12-01:
como faço pra conectar no wifi sem o lcd?