makerbase-mks/MKS-DLC32 #26: Info on v2.0 board

AussieMakerGeek, 2021-11-10:
Where is support files for v2 of board now shipping? schematic, firmware etc. please upload.

trashypopashy, 2021-11-10:
> Where is support files for v2 of board now shipping? schematic, firmware etc. please upload.

Where did You see a DLC32 V2?

@solawc @makerbase-mks @mks-viva are all the boards shipped in Version 2, now? Please show the differences!

AussieMakerGeek, 2021-11-10:
I purchased via the official makerbase Ali store but received one like this
https://www.aliexpress.com/item/1005003511460756.html

mks-viva, 2021-11-11:
@AussieMakerGeek I will upload it

trashypopashy, 2021-11-11:
Do I still have a TTL-Output-possibility on the V2-Boards?

mks-viva, 2021-11-11:
@trashypopashy 
Only 1 TTL has been reserved so far.

@AussieMakerGeek 
I has upload hardware sch and so on

trashypopashy, 2021-11-11:
> @trashypopashy Only 1 TTL has been reserved so far.

@mks-viva You mean there is only the CNC-Spindle-PWM-Output with its 12/24V?

mks-viva, 2021-11-11:
> > @trashypopashy Only 1 TTL has been reserved so far.
> 
> @mks-viva You mean there is only the CNC-Spindle-PWM-Output with its 12/24V?

Yes

AussieMakerGeek, 2021-11-11:
yes, but in the same as the 8bit version there is a ttl output before the fet so it does do 3v ttl, its just on the logic side of the spindle fet.

trashypopashy, 2021-11-11:
But there is no plugable interface anymore. I would need to solder, right?

AussieMakerGeek, 2021-11-11:
no, there is now a 3 pin ttl,gnd,vin connector plus separate vin,gnd connector. I am already using it for my diode laser.

mks-viva, 2021-11-11:
@AussieMakerGeek 
TTL is 5V output on dlc32 board
[image]

AussieMakerGeek, 2021-11-11:
@mks-viva thanks for all the info, can you briefly explain differences between 2.0 and 2.1?

mks-viva, 2021-11-11:
Refer to here: https://github.com/makerbase-mks/MKS-DLC32/tree/main/hardware
[image]

AussieMakerGeek, 2021-11-11:
sorry! I missed that bit. Thanks for clarification.

closed 2021-11-16 by mks-viva