makerbase-mks/MKS-DLC32 #27: update boart V1

GadgetX, 2021-11-11:
hi
any update to board V1 becouse for V2 don't work pwm on all the time :(

trashypopashy, 2021-11-11:
Please read what You wrote. I don't get the point in Your post.

Try to write complete sentences with "." and "," for better understandability.

GadgetX, 2021-11-11:
> Please read what You wrote. I don't get the point in Your post.
> 
> 
> 
> Try to write complete sentences with "." and "," for better understandability.

Now better??

trashypopashy, 2021-11-11:
Could Your question mean the following?

Is there an update for the V1.1-board? I am asking, because the V2-update doesn't work on my V1.1-board. When updating my V1.1-board, PWM is on all the time.

GadgetX, 2021-11-11:
> Is there an update for the V1.1-board? I am asking, because the V2-update doesn't work on my V1.1-board. When it is updated with the V2-firmware, my V1.1-boards PWM is always on.

Thanks 😉

GadgetX, 2021-11-18:
> > Could Your question mean the following?
> 
> > 
> 
> > Is there an update for the V1.1-board? I am asking, because the V2-update doesn't work on my V1.1-board. When it is updated with the V2-firmware, my V1.1-boards PWM is always on.
> 
> 
> 
> find it here
> 
> 
> 
> https://github.com/makerbase-mks/MKS-DLC32/tree/1044791e76b8b46b43c7ce75b30b6931f3bb4270?fbclid=IwAR0UoAcc6c__Np9v9iV99yPMdHFh9rlBjEQSz4yQrZJpKIpbYemHpBdtP-U
> 
> 

I ask about updated firmware to board v1.1 .

trashypopashy, 2021-11-18:
Well, ask @makerbase-mks @mks-viva @solawc ! We are all waiting for it.

GadgetX, 2021-11-20:
I see they don't support any more board v1.1🙄🙄🙄

trashypopashy, 2021-11-20:
If this will be true, they (@makerbase-mks @mks-viva @solawc) can be sure, I will get my money back (paypal), because there are still functions not working on my V1.1-board with the old firmware.

21mac, 2021-11-21:
> If this will be true, they (@makerbase-mks @mks-viva @solawc) can be sure, I will get my money back (paypal), because there are still functions not working on my V1.1-board with the old firmware.

you don't need to ask for the firmware 1.1
here it is:
https://github.com/makerbase-mks/MKS-DLC32/blob/c2451c41e7e60f6465e2da6f59000ffabc4d343f/firmware/Laser/Normal/DLC32_Normal_V1.2.001_8M_H35_0911.bin

GadgetX, 2021-11-21:
I ordered 2 numbers V2 boards and definitely I will ask refund for my 2 boards v1.1 because they don't support this board.I don't really understand how this is possible 2 months old board, and they ignored all problems with firmware 🙄🙄🙄

alfrankgit, 2021-11-22:
Why are the files for the V1 board hidden under that cryptic path that 21mac posted above?
Shouldn’t there be a clearly visible branch or even better a separate repository for that?
This is so confusing and frustrating!

alfrankgit, 2021-11-22:
Oh, and after installing this firmware onto my V1.1 board there is no WiFi configuration anymore.
When I click on the WiFi icon on the home screen, the screen goes all black.

trashypopashy, 2021-11-22:
From hardware-version 1.x to 2.x the, chabged some pins. So You better don't use firmware V2 for a V1.x-board.

alfrankgit, 2021-11-22:
Thanks!
But why is the V1.x firmware hidden?

GadgetX, 2021-11-22:
> Oh, and after installing this firmware onto my V1.1 board there is no WiFi configuration anymore.
> 
> When I click on the WiFi icon on the home screen, the screen goes all black.
Welcome to MKS World 😂😂😂

trashypopashy, 2021-11-22:
> Thanks! But why is the V1.x firmware hidden?

Just because the guys from Makerbase don't get it. I guess they don't care about logical documentation or customer support at all.

alfrankgit, 2021-11-22:
Point me to a Laser / CNC Controller with Display, SD-Card and a good UI with all necessary functions available now at a reasonable price and I'll forget MKS forever after years of being frustrated by them. ;-)

trashypopashy, 2021-11-22:
I would appreciate that aswell. By the way: Are You alfrank from drucktipps3d.de?

alfrankgit, 2021-11-22:
Yes, that's me.

trashypopashy, 2021-11-22:
Spitze, habe grad geemsehen, dass dort achon jemand mit dir textet. Ich steig da mal mit ein. MKS juckt es ja eh nicht, was hier für Probleme sind.

21mac, 2021-11-23:
> 

that's not cryptic, you just have to click on "History", since it is an "old" version.

closed 2021-11-23 by GadgetX