makerbase-mks/MKS-DLC32 #28: no cnc firmware in zip folder

Lazarus-Rising, 2021-11-11:
as it says, I downloaded the zip file for cnc normal and when I unzip it there's nothing there.
is a cnc version available?

trashypopashy, 2021-11-11:
This one is downloadable: https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/CNC/Normal

Lazarus-Rising, 2021-11-12:
Thanks but when I download it and extract using iZarc there is absolutely nothing in the folder!

trashypopashy, 2021-11-12:
There is no need to take the .zip-file. Just click on the link, posted above, click on the file-link again and then You have a download-button on the right. This way, You get the .bin-file itself.

Lazarus-Rising, 2021-11-12:
Hi, I've tried that and I can't find a download button. Please tell me I'm not senile just yet!
[image]

trashypopashy, 2021-11-12:
click on the yellow marked area:

[image]
-----------------------------------------------------------------------

...then...

-----------------------------------------------------------------------

[image]

Lazarus-Rising, 2021-11-12:
I'd done all that and it turns out it was file associations setting in IZarc!

Thanks for your help and patience!

closed 2021-11-12 by Lazarus-Rising