makerbase-mks/MKS-DLC32 #34: Web-Interface is missing some functions...

alfrankgit, 2021-11-13:
The web-interface is missing some important functions when compared to the touchscreen interface.
This is currently about the Laser Core XY firmware "DLC32_V2.0_D2.00.3_H35_8M_1108_C".

No labels (X-, X+, Y-, Y+) on the axis buttons.
No Z axis at all (also missing on the touchscreen).
No Unlock button.
No HHome button (whatever its functionality is).
No Laser power adjustment.

solawc, 2021-11-18:
thank you for your feedback

Ondrashek, 2021-11-28:
In firmware/TS35/Laser/Normal/Board_V2.0/version 2.0.4 on webUI -> SD File menu I am not able to close menu [marked with red circle].
Safari - iOS 15.1 - iPhone 13 Pro

[image]

abinoer, 2022-01-15:
@alfrankgit 
The functionality you proposed has been added, please update the firmware

closed 2022-01-25 by abinoer