makerbase-mks/MKS-DLC32 #35: dlc_cfg.txt update fails

alfrankgit, 2021-11-13:
Using the current firmware DLC32_V2.0_D2.00.3_H35_8M_1108_C.bin on the MKS DLC32 V1.1 board
the included dlc_cfg.txt update fails shortly before the end of the progress bar.

This is at the end of the file and it doesn't seem to match the firmware file:
/--Version Configrations-----//
//--Used to be displayed on About Menu-----//
//board_info_1
BoardName:MKS_DLC32
//board_info_2
BoardVersion:V2.0
//board_info_3
FirmwareVersion:V1.0

trashypopashy, 2021-11-15:
Maybe this could help You: #9

alfrankgit, 2021-11-22:
Nope. 
Where's the new cfg file that worked for you?

alfrankgit, 2021-11-22:
Oh, great, not the one from the hidden V1.x fork works but the V2.x one from here:
https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware

trashypopashy, 2021-11-22:
The only impprtant thing is, that You need to add $133, $134 and $135.

closed 2021-11-25 by makerbase-mks