makerbase-mks/MKS-DLC32 #36: What`s happen with Source Code release?

m3dkata, 2021-11-14:


solawc, 2021-11-18:
> _No description provided._

Do you want to ask what changes have been made to the source code?

trashypopashy, 2021-11-18:
I guess, the meaning is, that we all want the code to be open source.

m3dkata, 2021-11-23:
We all want the code to be open source! Issue #10 
[image]

bnorth12, 2021-12-17:
So is MKS-DLC32 a unique code base or s it based on previous open source efforts?

bnorth12, 2021-12-17:
Is MKS DLC32 based on or branched from GRBL_ESP32? Getting the PIN definitions assigned would be good.

RandomHacks-Git, 2021-12-17:
> Is MKS DLC32 based on or branched from GRBL_ESP32? Getting the PIN definitions assigned would be good.

You can check the pins for whatever board version you have on the [hardware folder](https://github.com/makerbase-mks/MKS-DLC32/tree/main/hardware), it's the pin.pdf file

capi92, 2022-02-11:
Any update on this ?

0727fqh, 2022-02-23:
When will open source DLC32? MKS-DLC32 is a modified code based on Bdring/Grbl_Esp32, right?

simulz, 2022-04-06:
source code please~

serial-tintypist, 2022-04-06:
I would also like to know when the source code will be released (as the explicit promise of it becoming open source was an important part of my decision to switch over to MKS DLC32 in the first place).

jerhole, 2022-04-14:
Was thinking on buying this controller, as I've went with mks on all of my printers. Though it seems there updates to older products is non-existent. Still waiting on laser support (or at the very least, instructions) from the nano V3. With all of the issues I'm reading on, from people that have purchased this controller,and the lack of follow through of open sourcing (what sounds to be a derivative of open source software, so should've been in the first place}. I'm about fed up with mks desire to pump out products half cocked. Still can't use mks WiFi with standard marlin neither. Get a grip. Finish the products and importantly the documentation of stuff youre already filling the landfills with. Why not focus on a handful of really good products, with good user experience, and make a good reputation that way. Also, while trying to not mass produce so much stuff, to add to the massive pollution problem we have. Money is gonna do nothing when this planet ceases to sustain us. Be a part of the solution, not greedy capitalistic cut throats

makerbase-mks, 2022-11-15:
Code open source late but finally here: https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

closed 2022-11-15 by makerbase-mks