makerbase-mks/MKS-DLC32 #37: MKS DLC32 V2.0

una2012, 2021-11-15:
Hello, I have a question about MKS DLC32 V2.0 board. Can I connect to wifi if I don't have a tft display

I have this message

VER:1.1.2021051202:]
[OPT:PHSW]
[MSG:Using machine:MKS DLC32]
[MSG:No Wifi]

21mac, 2021-11-18:
yes, that is possible.
Just connect your controller board to your computer, open any laser engraving software (or any other that can send gcode commands) like "Lightburn" or "LaserGRBL" and send the commands from this documentation:
https://github.com/bdring/Grbl_Esp32/wiki/Settings

I used laserGRBL and entered all commands in this text box, press the "return" or "enter" key to confirm each command
[image]

the commands where:
[ESP100]MyWifiSSID
[ESP101]MyWifiPassword
[ESP115]ON

Attention: ESP115 turns the radio on TEMPORARILY, after a reboot the radio is off again.
use 
[ESP110]AP
to set the mode to "access point" so the ESP32 connects automatically after a reboot

makerbase-mks, 2021-11-25:
@una2012 , yes, you can, please update the latest firmware on our github, you can config the wifi infomation through the config file.

closed 2021-11-25 by makerbase-mks