makerbase-mks/MKS-DLC32 #38: MKDLC32 TMC2208 TMC 2209 Compatible

Devil-67, 2021-11-15:
Unfortunately I could not find the answer on the net.
I have just ordered the MKS DLC 32 from Ali and am reluctant to use the enclosed 4988 drivers because of the low resolution.

is the MKDLC32 
TMC2208 Compatible, if so which version V1.2 / V2 or V3? 
or do the 2209 also work ?
Or in other words which one would you recommend.
Thanks
Mario

solawc, 2021-11-18:
> Unfortunately I could not find the answer on the net. I have just ordered the MKS DLC 32 from Ali and am reluctant to use the enclosed 4988 drivers because of the low resolution.
>
> is the MKDLC32 [...]

In fact, 4988, 2208, and 2209 are all available. Of course I recommend using 4988 or 2209. In my own experience, the effect of 2208 is not very satisfactory.

MiroHibler, 2021-11-18:
> In fact, 4988, 2208, and 2209 are all available. Of course I recommend using 4988 or 2209. In my own experience, the effect of 2208 is not very satisfactory.

What are the advantages of 2209 over 2208 from your experience? I plan to use this board and 2208s for the CNC 3018 Pro, so I did a quick test and it seems to be working okay compared to the original Woodpecker board with 4988s (minus the Z axis which is still dead in the [latest FW](https://github.com/makerbase-mks/MKS-DLC32/blob/main/firmware/CNC/Normal/DLC32_CNC_V1.2.001_Normal_8M_H35_1009.bin) 👎 ).

Are there any parameters to be adjusted when moving from 4899 to 220X?

matejc88, 2021-11-18:
I installed TMC2225 and it is working. I only dont know if i set voltage right but looks ok.

Devil-67, 2021-11-18:
I find the TMC 2209 very interesting.
However, it is not clear to me what V1.2 or V3 is all about and which would be the better option.
4899 / 2208 / 2209
I would like my laser cutter to drive smoothly and also to have a high acceleration because it is very big with 1.5m and the paths are long.

@solawc  you think the 4899 is better than the 2208 ?? that amazes me actually it should be the other way round ?
Or have I misunderstood something ?

makerbase-mks, 2021-11-19:
@Devil-67 The rated current of TMC2209 is larger than TMC2208, so the torque is also larger. solawc mentioned that the effect of TMC2208 is not as good as that of A4988, which also refers to the torque. In terms of silence, the TMC series is definitely better than A4988.

Plottmania, 2021-11-21:
I do not understand!
With 17er engines it does not matter what there is a driver on the board!
True, the TMC are much quieter and also run much better!
But if I connect two motors on the Y-axis, the two motors now share a driver!
Now think well where the journey goes : )
Now there are several options such as external drivers.
In addition, an adapter from Barton Dring to the DLC and there no matter which...already all drivers can be connected.
Just pay attention to the connections.
The Y-axis 2 is rotated (cloned).
The TMC 2208 V3 are very good for small lasers with two axes for example and are completely sufficient.
And no, no matter if there are any jumpers plugged in or not...the resolution of the steps remains at 800 in the GRBL.
So simple is cinema....

https://plottmania.de/article/9-mks-dlc-2-0/

[image]

closed 2021-11-25 by makerbase-mks