makerbase-mks/MKS-DLC32 #39: Fan output

Anderson8153, 2021-11-16:
How could it be done to control a fan in place of the Z axis or other output?

21mac, 2021-11-18:
to control it via Z axis output is quite difficult since it's not just a power output pin but a controller for a bldc motor.
Why don't you just plug the fan into the 12 Volt port? so the fan would be on full speed as long as the board is turned on.

There is no "easy" solution to control the fans speed via an output pin.

Anderson8153, 2021-11-19:
@21mac , Hi, I understand, I just want to turn the fan on and off via Gcode

21mac, 2021-11-19:
I don't think that will work.
Since - as I meantioned- there is not output pin (from the microcontroller ESP32) which connects to a controllable pin for the fan, there is not Gcode for it.

The board only has one ttl (= PWM) port which is reserved for the laser.
The other ports/sockets are for the motors, the spindle, the end switches or are just power pins (12V, 24V).

There is not possibility to turn these ports on/off.

only in one scenario it will work: if you dont use the ttl (PWM) port for a laser.
- which board (1.0, 1.1, 2.0, 2.1) do you have?
- do you use a laser module or a cnc spindle motor?
- how did you connect everything? can you upload a picture of your board with all cables connected?

Anderson8153, 2021-11-19:
@21mac , 2.0 board with laser module, connected to TTL output.

I connected the fan to one of the 12/24V outputs and installed a switch

[image]
[image]
[image]

21mac, 2021-11-20:
> @21mac , 2.0 board with laser module, connected to TTL output.
> 
> I connected the fan to one of the 12/24V outputs and installed a switch
> 
> [image] [image] [image]

Damn. there is no other software / Gcode controllable port, just the ttl.
Since the firmware is not open source there is no way to use another pin of the ESP32.

sorry

Anderson8153, 2021-11-20:
@21mac , Don't worry, thanks for the help!

makerbase-mks, 2021-11-23:
Sorry, as 21mac said, now there is no method.

closed 2021-11-23 by makerbase-mks