makerbase-mks/MKS-DLC32 #40: How to change Home screen photo?

Anderson8153, 2021-11-16:
How to change Home screen photo?

solawc, 2021-11-18:
> How to change Home screen photo?

Regarding the picture, there is no replaceable form on DLC32, because we compile the picture into source code and save it, which can be modified after open source.

makerbase-mks, 2021-11-19:
@Anderson8153 Just now, you can send to us to help you to change.

Anderson8153, 2021-11-19:
@makerbase-mks,
[image]
 Thank you very much, I will send the file

Anderson8153, 2021-11-19:
@makerbase-mks What software do you use to compile the firmware

makerbase-mks, 2021-11-23:
@Anderson8153 Sorry for delay, may I know :
1. Your board version, V1.x or V2.x
2. Your machine structure, Normal XY or CoreXY
3. Laser or CNC?

Anderson8153, 2021-11-23:
@makerbase-mks Hello, no problem!

V2.0  laser normal

makerbase-mks, 2021-11-25:
[DLC32_V2.0_D2.004_H35_1123_N_change_logo.zip](https://github.com/makerbase-mks/MKS-DLC32/files/7600070/DLC32_V2.0_D2.004_H35_1123_N_change_logo.zip)
@Anderson8153 , please have a try.

closed 2021-11-25 by makerbase-mks

Anderson8153, 2021-11-25:
@makerbase-mks,
At first there is only one problem!
when I give the Frame command on the GRBLlaser the laser doesn't turn on, but it works normal offline

Anderson8153, 2021-11-25:
@makerbase-mks, The laser fan is always on too, in the latest version that was released on the forum too!

Ams-Laser, 2023-03-24:
can you please helpme change my logo MKS-dlc32 V2.1 laser firmware V2.30(8M.H35.20221223)
[image]