makerbase-mks/MKS-DLC32 #42: ts35-r v2

kweilim98, 2021-11-17:
can i have ts35-r touch screen panel 3d cad file?

solawc, 2021-11-18:
> can i have ts35-r touch screen panel 3d cad file?

I don’t quite understand what you mean, but I can describe it more clearly

21mac, 2021-11-18:
he asked for a 3D model of the touch screen, as there is a 3D file (.step) of the MKS-DLC32 as well.

keweilim98, do you need the 3D model for constructing anything?
i did not model the touch screen but I made a housing for the DLC32 and the TS35 in blender.
if you like I could upload the .stl and .blend to thingiverse.

makerbase-mks, 2021-11-19:
@kweilim98 Will this help you? https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20TS35-R.DXF

kweilim98, 2021-11-19:
Yes, I need the step file

kweilim98, 2021-11-19:
> he asked for a 3D model of the touch screen, as there is a 3D file (.step) of the MKS-DLC32 as well.
>
> keweilim98, do you need the 3D model for constructing anything? i did not model the touch screen [...]

YES , I need the 3d step file for that panel

21mac, 2021-11-19:
> YES , I need the 3d step file for that panel

ok. sorry, I don't have a model of the LCD. I just made 3D models of a housing for the board and for the LCD.
Aside from that I use blender and can't export to .step (just .blend, .stl, .fbx. ,obj etc.)

Ondrashek, 2021-11-23:
> ok. sorry, I don't have a model of the LCD. I just made 3D models of a housing for the board and for the LCD. Aside from that I use blender and can't export to .step (just .blend, .stl, .fbx. ,obj [...]

Hi,

Can you upload your housing on thingiverse? I am interest. Thanks.

21mac, 2021-11-23:
for anyone who needs it:
I made a crude housing for the board and the display: 
https://www.thingiverse.com/thing:5144048/comments#comment-6226596

makerbase-mks, 2021-11-25:
> YES , I need the 3d step file for that panel

Sorry, we only has the dxf file for the panel.

closed 2021-11-25 by makerbase-mks