makerbase-mks/MKS-DLC32 #43: Usage of "switch" on MKS-DLC32 V2.0

21mac, 2021-11-17:
Hi,
I wondered what the usage of the "switch" pins is. They are labelled as "J3" and there is this diagram in the documents:
[image]

So basically I can solder a switch to these pins to bypass the zener diode, correct?
What would be the purpose of doing so?

solawc, 2021-11-18:
> Hi, I wondered what the usage of the "switch" pins is. They are labelled as "J3" and there is this diagram in the documents: [image]
> 
> So basically I can solder a switch to these pins to bypass the zener diode, correct? What would be the purpose of doing so?

This position is convenient for users to use external switches to control whether their machines need to be powered on or off. As you said, you can bypass the diode externally

21mac, 2021-11-18:
Thanks for your reply.
First of all: I mistook the symbol, it's not a zener diode but a "polyfuse" (the plug-in-fuse on the board).

But bypassing it would not turn the board on or off.
When I plug the round connector into the socket, then the board instantly turns on, but the switch (which is not soldered onto the board yet) is "normally open".
So what would happen if I closed the switch? Nothing. the board IS ALREADY on and it would stay on.

The only way of using a switch that I can imagine: Removing the fuse, so that there is no other way to travel the circuit but through the switch.
But then I won't have a fuse on the board and I had to solder one into series with my switch?!
And I don't want to push 12 volts and several amps through a switch that I touch.

I am no specialist in electronis but I thought I could read the schematic/diagram correctly.
The image I posted earlier does not make any sense to me.

makerbase-mks, 2021-11-19:
> Thanks for your reply. First of all: I mistook the symbol, it's not a zener diode but a "polyfuse" (the plug-in-fuse on the board).
>
> But bypassing it would not turn the board on or off. When I plug [...]

@21mac , Yes, you are right, the way of using a switch is to  remove the fuse, and you should choose a switch which can load the current of the hold board(Suggest larger than 5A).

21mac, 2021-11-19:
> @21mac , Yes, you are right, the way of using a switch is to remove the fuse, and you should choose a switch which can load the current of the hold board(Suggest larger than 5A).

Thanks a lot, now I understand :)

closed 2021-11-19 by 21mac