makerbase-mks/MKS-DLC32 #44: 3d probe and tool probe

kweilim98, 2021-11-20:
Does this motherboard able to make 3d probe and tool probe together for CNC router since I saw the board got only one port to connecting

makerbase-mks, 2021-11-25:
Sorry, i don't understand what you mean

kweilim98, 2021-11-25:
> Sorry, i don't understand what you mean

I want to know how the connection for the 3d probe and the z-axis tool probe with the motherboard MKS dlc32 for CNC router since for CNC user with this board, we need two functions of this probe to setting the zero value for the homing position

makerbase-mks, 2021-11-25:
@kweilim98 I know z-axis tool probe like this: https://www.aliexpress.com/item/1005001490059077.html?spm=a2g0o.productlist.0.0.665c3aef7L6bHR&algo_pvid=7211cd1a-4597-44a1-ba9e-d43ea7e5f784&algo_exp_id=7211cd1a-4597-44a1-ba9e-d43ea7e5f784-1&pdp_ext_f=%7B%22sku_id%22%3A%2212000016321420747%22%7D

But what is the 3d probe, just another sensor but the same usage?

kweilim98, 2021-11-25:
> @kweilim98 I know z-axis tool probe like this: [...]

[image]
3d probe like this

makerbase-mks, 2021-11-29:
I will ask engineer to enable z-axis tool probe first.

makerbase-mks, 2021-12-15:
You can use the lastest released firmware to use tool probe now.

kweilim98, 2022-05-24:
When the X y and z axis probe features will be added? Now the CNC firmware only have the z probe only

corpix, 2025-10-15:
Not sure if this solved or not, but I've got the same probe for my TTC450 (based on makerbase DLC32 v2.0).
Looks like probe could be connected to 3 pin connector, but it requires [inverting sensor pin](https://github.com/makerbase-mks/MKS-DLC32/blob/856c74f19d1f947dd3123bd3f5a64ad4bd26ea56/MKS-DLC32-main/firmware/dlc_cfg.txt#L29) because probe IO pin will "pull up" a sensor pin when triggered (while default behavior for sensor pin on the board is "pull down")

corpix, 2025-10-15:
Got this probe connected and working according to PIN diagram for v2.1 

[image]

Here you can see a probe triggered by hand in bCNC controller
[image]

Only thing I need to do was `$6=0` to invert the probe