makerbase-mks/MKS-DLC32 #45: Frustrated and a few questions

Plottmania, 2021-11-20:
Hello,
I have been dealing with products from MakerBase for quite some time : )
Especially the MKS DLC has done it to me but now I have to ask!
My DLC 32 V1.1 (so it is shown in the display) works via the WebUI but when I start this I have the feeling that I can not do everything!
Bad enough if you have to enter the WiFi password after each restart of the DLC....ich know there other very well functioning systems such as a xPro V5 just as an example.
Now should come an UpDate but what comes?
A version for the MKS DLC 32 V2.0 which I have of course also tried times....Horror!
To the understanding, I have here a diode laser at a X-Y system (no core) and thus per axis a motor.
I can use LightBurn without problems via USB, LaserGRBL without problems, use the display which works fine for now but the WebUI is a kick in the butt.
If I now read various answers to questions regarding updates I would have to puke!
One has the feeling MakerBase has no desire for support but is horny for profit.
I have worked my butt off to explain to people how they have to set what and where.
My documentation is much better than the MakerBase documentation on the MKS DLC...fact.
I've had the DLC 2.0, the DLC 2.1, the XY DLC and now a DLC32 V1.0 here for two weeks to realize I bought a crutch?
No, I also still buy MKS TMC 2160 and other products...there MakerBase has done everything right : )

I would like to see a proper documentation of the hardware, maintain the software and not throw half-finished products on the market.

Yes, I'm very pissed off because like every noob I bet on a simple and easy to use product which turns out to be a gravedigger.

The question is quite simple and should please be answered professionally!
Is the WebUI for the DLC 32 V1 still extended or come the desired features?
Is there a solution regarding the password and the question of all questions whether there is also a documentation of the hardware?
Here times as a hint the question of what the jumpers on the V1. are or the dips on the V2.....mir is known for what because I played around with it for days!

Do your homework and do not leave people hanging here!!!
For my bad English I would like to apologize, my German is a little better but the few understand here!

My text was translated with Deepl....Thanks

21mac, 2021-11-20:
what especially is your problem?
You can connect to the webUI, you press a button (like "move X") and the laser does not do anything?
And it works with laserGRBL and lighturn?

you could copy the Gcodes that the webUI outputs and put them manually in laserGRBL (https://lasergrbl.com/usage/user-interface/  see #3).
does it work?

could you post the full (verbose) output of the webUI please (attach it as text file please).

I got the 1.1 and now the 2.0 and both work fine over wifi/webUI.
To exclude problems with your wifi network, you could open a hotspot on your smartphone and connect the MKS32 directly to the hotspot wifi.

did you try to set your wifi settings via ESP commands (ESP100, ESP101, ESP110, ESP115) instead of using the touch screen?
you find the documentation [here](https://github.com/bdring/Grbl_Esp32/wiki/Settings).

Plottmania, 2021-11-21:
Well, really understood what I wrote?
Really read and really understood my questions?
I am not alone with my questions but I ask again now.

Where is the difference between the V1.0 and the 2.0
Why are not all features available in the WebUI for my V1.0 and why is the statement that these are still being implemented.
You know what documentation is?
Something like this is not limited to links where it is explained how to build your network.
My question why I lose the password and I do not need options to anything.
Well, I have the whole day to deal with it again but if I buy something then I want to use the options that are described.
Options that are also feasible!
Well, I think with my approx. 30 GRBL controllers and meanwhile 7 laser modules and various machines I have enough experience and there are not really many secrets.
What I do not do like many other users also...to deal with firmware!
The comes from the provider and should funktioinieren or?
Right, but it does not work properly!
Since I look at the xPro V5, the controllers of Barton..... goes now a light?

Translated with www.DeepL.com/Translator (free version)

21mac, 2021-11-21:
Since you write things like "it should funktionieren":
ich versuchs mal auf Deutsch:

Ja, als Softwareentwickler und nach einigen Jahren Erfahrung weiß ich sehr gut, was eine Doku ist, wie man sie schreibt, wie man mit solchen Problemen umgeht.
Und da ich auch mehrere Jahre "Head of support" war, versuche ich nun, mich an dein Problem ran zu tasten.

Dass das Webinterface nicht alles kann ist doof, aber ändern kannst du es nicht. Ich meine, wir sprechen hier über einen 30 Euro Controller aus China. Wenn es dir wichtig ist an der Firmware was zu ändern, an der Webseite oder neue Features zu implementieren, dann nimmt dir GRBL und pack es selbst auf einen Arduino oder ESP32.

Du hast eine Version 1.1 gekauft und im Vergleich dazu ist die 2.0 schon viel besser. Makerspace arbeitet doch dauernd an neuen Versionen aber das WebUI halt halt keine so hohe Priorität, wie ich das sehe.
Und dass man neue Features auf einem "alten" Board nicht nachträglich hinzufügen kann ist auch klar, oder?

"why is the statement that these are still being implemented.".
Das liegt daran, dass es bei Makerspace wohl keine hunderte Mitarbeiter gibt die sofort auf alle Anfragen reagieren können.
Als Softwareentwickler weiß ich, dass Anpassungen oder neue Features nach Priorität gemacht werden und gerne mal mehrere Monate dauern können.

Wieso brauchst du auch unbedingt das WebUI? Gravierst du von der SD-Karte aus?

Was ich eben NICHT verstanden habe: Hast du neben dem 1.1 noch ein zweites BOARD 2.0 gekauft oder hast du die Firmware 2.0 auf das Board 1.1 drauf gemacht?

Nein, es gibt keine Releasenotes, wo du direkt 1.1 und 2.0 vergleichen kannst. Und nachgucken kannst du auch nicht weil es nicht open source ist.

Das Produkt ist halt "as is" und ende. Ob das gut ist oder nicht, das musst du wissen.
Wenn es dir nicht taugt musst du es eben zurück senden und dich nach einem anderen Controller umsehen.

Ich benutze alle Laser/CNCs an einem PC, mit LaserGRBL, Lightburn etc.
Da passt alles und das WebUI ist mir egal

Plottmania, 2021-11-21:
Hallo und Danke für Deine Antwort.
Ich schreibe in Deutsch weil Dein Deutsch für mich perfekt ist und verstehe wo die Reise hingeht.
Die Reise geht ins Nirvana und wenn Du wirklich im Support gearbeitet hast und Dich gut auskennst meine Frage ob es normal ist Leuten vor die Stirn zu kloppen?

Ach, ich habe es nicht verstanden und Dir ist egal ob ich da eine WebUI nutzen möchte?
Die WebUI wurde so beworben und ich möchte nicht wissen wie viele von den Teilen verkauft wurden!
Still, Klamm und Heimlich wurde geswitcht und ab sofort ist der MKS DLC 1.0 nur noch ein Abfallprodukt?

Entschuldigung....Support, Informationen und Dokumentationen sehen für mich ganz anders aus.
Nochmal, weder juckt mich Deine Erfahrung, Deine Erkenntnis sondern einzig die Frage wie es machbar ist ohne im GRBL, einer Config.h oder was auch immer zu schrauben damit etwas funktioniert?

Dann hätte man dies mitteilen können und wie Du siehst ist auch zum DLC 2.0 keine richtige und eindeutige Information und Dokumentation vorhanden.
Die Aussage ist jetzt besser und nur ein Abfallprodukt von Abfall.

Persönlich und dies konntest Du erlesen habe ich einen DLC 1.0 und habe die Version für den DLC 2.0 geflasht.
Wäre ich jetzt nicht so höflich und freundlich hätte ich geschrieben MakerBase kann sich den Scheiß in den Hintern stecken.
Ich habe so ziemlich alles an MakerBase-Produkten was den MKS DLC betrifft und würde ich solch ein Produkt auf den Markt bringen bestimmt sehr viel durchdachter.

Wir sind hier bei: billig nachschieben!

Mir muss niemand erzählen was eine Dokumentation ist, ein funktionierendes System und wenn ich schon nicht in der Lage bin etwas sauber zu programmieren gibt man dies der Masse.
OpenSource ist hier nichts und nochmal, ich hab ein Produkt gekauft was nicht mal zu 50 % fertig ist oder eines gekauft was nur zu 50 % fertig gestellt wurde.

Den Rest darf das Volk machen aber wie?

Meine Aufgabe ist es bestimmt nicht irgendwo in der Firmware zu wurschteln!
Deine Aufgabe ist es nicht mir mitzuteilen ich sollte die WebUI einfach nicht nutzen.

Einer der Kaufgründe war diese Option!
Wie man damit umgeht sehr ich gerade.

Es ist nur meine Meinung aber ich wette damit stehe ich nicht allein!
Ja, ich nutze nur LightBurn und zum flashen der DLC 2.0 /2.1 und Co normal LaserGRBL.
Für mich ist es Spielerei da ich hier über 30 GRBL-Controller habe, diverse Systeme und das Ganze als Hobby betreibe.
Nochmal, angepisst wenn mir dann und seien es nur 20 € für Schrott aus der Tasche gezogen werden!
Ja, ich hab mich breitschlagen lassen zwei DLC32 V2.0 zu bestellen.

Flashen geht ja prima und etwas anders als hier beschrieben und verlinkt....wir sind bei der Dokumentation die nach wie vor sehr mies ist.

So, ich habe meine Meinung geteilt und keinen Frust.
Ich habe mich damit abgefunden ein zu 50 % funktionierendes Produkt gekauft zu haben und festgestellt, Support beherrscht nicht jeder.
Für mich ist die Kaufempfehlung was MakerBase-Produkte betrifft mittlerweile mit Vorsicht zu genießen.
Ist aber auch egal denn einige Produkte wie die TMC 2160 oder Servos funktionieren auch an anderen Controllern von mir.
Ach ja, es hat schon einiges gedauert bezüglich der Servos und wie diese eingestellt werden.....12 Monate?

Danke und einen schönen Sonntag

21mac, 2021-11-23:
Ja, das habe ich verstanden.
Aber wie immer - sagte ich ja bereits - wenn dir das Produkt nicht gefällt schick es zurück.
Wenn du aber schon vorher keine Doku etc. gefunden hast, wieso hast du es dann bestellt?

Doch, open source würde hier helfen. Dann könnte die Community oder man selbst vieles anpassen oder neu implementieren.

Du tust gerade so, als hättest du tausende Euro für eine Industriemaschine bezahlt. Ein Teil für... 20 Euro?... funzt net so wie du willst. Dann schick es zurück oder schreib es ab. Ich hab ja nichts **dagegen** gesagt, dass du dein Geld zurück verlangst. Das steht dir ja zu.

Dass dich meine Erfahrung nicht juckt ist mir auch egal, aber DU fragtest, ob ich weiß was Doku etc. sind.
ICH schreibe meine, genau genommen habe ich eine Doku für eine zehntausende Euro teure Software geschrieben, den Support mache ich auch anders als MKS.

Nicht jede Software ist gut, nicht jedes Produkt funzt wie gewohnt.
Du musst dich nicht selbst mit der Firmware rumschlagen, wenn das nicht willst (oder im Fall von closed source: nicht kannst).

makerbase-mks, 2021-11-23:
@Plottmania Sorry for delay. 

The webUI is available. I don’t know exactly what kind of problem you encountered?

You mentioned that another question should be that the wifi password needs to be re-entered after the power is turned on again? This problem has been fixed in the two versions of https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TFT35/Laser/Normal/Board_V2.0, of course, it is recommended to use the latest compilation version of.

Plottmania, 2021-11-23:
Hallo 21mac,
ich teile ganz bestimmt vieles und einiges Deiner Meinungen und Einstellungen.
Was die DLC betrifft und ich hab da wirklich einiges oder besser gesagt alles : )
Ich betreibe das ganze als Hobby und ichhabe hier schon weit schlechtere Controller auf dem Tisch gehabt.
Sie alle funtionieren mit LightBurn und LaserGRBL und zu fast keinem gab es irgendwo ausser Pinbelkegungen eine Anleitung.
Gut, einem WoodPecker 3.4 und vertauschten Endschaltern....ich kenne die Problematik.
Es war auch kein Angriff gegen Dich nurhab ich in diversen Foren diesen MKS DLC32 beschrieben wie Sauerbier und wie geil es ist...ist es im Grunde ja auch nur kotzt es einen an wie hier klammheimlich der Kunde und da bin ich nicht allein verarscht wird.

Nach wie vor stehe ich zu den Produkten von MAkerBase aber wieso werden unterschiedliche Displays ausgeliefert, kein Reset und andere Kleinigkeiten?
Also für mich ist "durchdacht" wenn überhaupt der XY DLC weil dieser zumindest einen ON/OFF und Reset hat.
Jedes halbwegs vernünftige Board hat solch ein Feature was ich sogar als sehr wichtig erachte!

Open Source, würde ich gern betreiben nur hab ich davon keinen Plan.
Schon grausam wenn nicht mal die MKS Software - Tool online für den DLC geschaffen wurde.
Fast jedes Board aber die Fraktion der DLC und hier die Jungs wie mich mit nur X/Y Achsen vollkommen aussen vor sind.

Da lobe ich mir einen xPro V5 wenn auch eine ganz andere Preisliga.
Was ich mit den DLC`s mach kannst Du Dir hier anschauen...nur Spielkram aber der funktioniert!

https://plottmania.de/forum/board/39-makerbase-dlc/

Ich bau zurück, guck was da mit FluidNC geht und versuche dies zu verstehen.
Wenn es eine Option ist sehr gern und der DLC32 V1.0 muss nicht in die Tonne wandern oder besser gesagt ich könnte ihn dann doch voll nutzen....für den Moment würde ein DLC 2.0 / DLC 2.1 oder XY DLC reichen.

Danke nochmals für Deine Meinung und bitte nicht falsch verstehen...ich habe gesagt die Baureihen der DLC gefressen und würde behaupten sie hätten für mich kein Geheimnis mehr.
Ein rotes Tuch wenn solch ein System (die Hardware) als Open Source angeboten wird aber es nicht ist.
Wir sind hier als Anwender und nicht als Programmierer und wollen ein funktionierendes System und keine Bastelbutze.
Bis zu einem gewissen Grad kann man vieles erlesn und selber machen aber mit anschliessen, starten und ein GRBL drauf klatschen ist es nicht getan oder nicht richtig.

Ich kaufe auch nur ungern ein Auto mit 6 Gängen von denen ich eventuell nur 5 Gänge nutzen kann...wo ist da also der Unterschied außer preislich?
Für mich eine falsche Denkweise zu behaupten ich könne es indie Tonne treten oder zurückschicken.

Na, mal probiert mit dem Verkäufer "MakerBase" über AliExpress?
Das ist dann noch mal sehr viel mehr Kino in dem ein schlechter Film läuft...mono, schwarz/weiß und abgedroschene Sitze.
Nicht mal Eis gibt es in der Pause....nene

Plottmania, 2021-11-23:
> @Plottmania Sorry for delay.
>
> The webUI is available. I don’t know exactly what kind of problem you encountered?
>
> You mentioned that another question should be that the wifi password needs to be [...]

Sorry but the question was quite different!
Since the problem with the lost password would be the smallest!
Your WebUi is lousy and we are not all options available as in the display.
If I do something wrong here please explain very briefly how I can do it better.
Compared to my xPro V5, the WebUi here looks quite modest....useless!
Meanwhile, I'm not the only one who complains about this.
I would have said you have not done your homework.
When I look at what I'm tinkering with your products and still make the effort here in Germany to help people who have a DLC ... I should really give it up!
I say goodbye to a WebUI and clap via USB again the cucumber controllers to the PC...plug in, turn on and runs.

I've pretty much slapped every update on the DLC32 V1.0 but unfortunately I don't know Chinese.
Apart from that I think you have not really understood the whole problem or?

Regards Stephan

Plottmania, 2021-11-23:
I just see there was an update for the DLC32 V 1.0 (Laser) just 5 minutes ago
I will try and report.

Thanks

https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/Laser/Normal/Board_V1.0

makerbase-mks, 2021-11-23:
@Plottmania Thanks for your advice! But I found the function on Barton's is similar with ours. I mean functions, of cause, it has some different details . And we also hope you can put forward more valuable opinions

Plottmania, 2021-11-23:
Hello,
thank you very much for the feedback.
It is quite simple as I use a laser and a system with two axes whereas I also have systems with three axes.
The WebUi for these systems is quite simple and if I use the WebUI via my tablet basically a large handwheel would be....the desire of many: )
There belong then on a side functions like:
- Start/Stop/Pause
- homing
- Zero point
- and if it runs well a parking point
- Override for speed and power
You can display this beautifully on a 7 inch display and I bet these options are the desire of many.
If now still the autonmatic connect is solved (possibly already solved) you have a lot of fun.
Then I take my hat off!!!

Thanks again for the replies.

Translated with www.DeepL.com/Translator (free version)

Plottmania, 2021-11-23:
a small update after the update...
At least the automatic connection works.
Thanks for that and hope one takes care of my wishes now for the rest....the main reason

makerbase-mks, 2021-11-25:
> Hello, thank you very much for the feedback. It is quite simple as I use a laser and a system with two axes whereas I also have systems with three axes. The WebUi for these systems is quite simple and if I use the WebUI via my tablet basically a large handwheel would be....the desire of many: ) There belong then on a side functions like:
> 
> * Start/Stop/Pause
> * homing
> * Zero point
> * and if it runs well a parking point
> * Override for speed and power
> * Laserpower ON/OFF (Settings for Power 0 -20 % for alignment)
> 
> A dream would be even: Enable/disable Z-axis
> 
> You can display this beautifully on a 7 inch display and I bet these options are the desire of many. If now still the autonmatic connect is solved (possibly already solved) you have a lot of fun. Then I take my hat off!!!
> 
> This system, so set and easy to set..I claim a best seller would be! I do not want to tell you now how many I know who have switched to a DLC because it is cruel with various programs and tools and controllers from various vendors.
> 
> Thanks again for the replies.
> 
> Translated with [www.DeepL.com/Translator](http://www.DeepL.com/Translator) (free version)

Thanks for your good advices again! In response to the suggestions you listed, we have confirmed:
 * Start/Stop/Pause  ———— Done
 * homing ———— Done
 * Zero point ———— Done
 * Runs well a parking point ———— TODO
 * Override for speed and power ———— Done (Can be config speed and power during engraving)
 * Laserpower ON/OFF (Settings for Power 0 -20 % for alignment) —— TODO
 * Enable/disable Z-axis ———— To be evaluated(Now the latest firmware enable the Z axis defaultly, it is necessary to be on/off?)

Plottmania, 2021-12-15:
It's more of a gimmick for me because I have multiple systems.
But now I want to mount my DLC32 on all machines.
I have 2-axis systems and a milling machine with 3 axes.
Sorry for the late feedback, I am working a bit in between.

I will try both versions of the DLC (V1.0 with display and V2.0 via WebUI) this weekend.

Thanks for the additions and a feedback from me if it works comes.

Thumbs up...

Markus-K40, 2022-08-31:
I tried to update the firmware of my DLC32 V1.0 but no update worked and nothing happend. The actual firmware doesnt support a z-axis and thats what I need for the motorised laserbed. 
I hope anyone can give me an idea.
Thanks a lot.

21mac, 2022-08-31:
> I tried to update the firmware of my DLC32 V1.0 but no update worked and nothing happend. The actual firmware doesnt support a z-axis and thats what I need for the motorised laserbed. I hope anyone can give me an idea. Thanks a lot.

one very polite question:
Do you think we can read minds or see through our magical glass spheres?

"no update worked and nothing happened" is not very helpful.
Do you tell the mechanic "my car does not work"?
"what exactelly? how does this show?"
"i dunno, it does not work".

how did you update it? Tell us step by step.
"nothing" happened? while updating or when powering up the DLC32? Do you have a display attached to it? which one?
Could you connect the webinterface afterwards?

which firmware did you use? be aware that you can ONLY use the firmware marked as "board_v1.0".
"board_v2.0" does NOT work in boards of version 1, obviously. 

which configuration file? did you do any changes?

the more you tell us the better we can help you.
please don't make people ask again and again for more details.
YOU have to deliver, else we can't help you.

Thank you

Markus-K40, 2022-08-31:
well here are the details. I saved the bin-file to the sd-card, put it in the slot and turned on the board. But the board starts normally without any update. I tried it with or without the MKS config-file and any bin-file for V1.0 but nothing. 

I did all changes from Lightburn and this worked fine. I used the basic-config because no file was accepted.

The webinterface is also working but not really helpful for z-axis. X and Y ist fine on the display and the web-interface but no z-axis at all.

where can I get the installed versionnumber?

the board is working and starting but not updating. The display is a TFT35 V2.0-R

Plottmania, 2022-09-01:
Markus, es bringt Dich nicht weiter weil auf der anderen Seite jemand sitzt mit immer den gleichen Fragen.
Statt die Firmware zu dem machen was sie hier sein sollte: OpenSource gängelt man!
Du kannst auf dem Display sehen welche Version auf Deinem DLC32 geflasht wurde.
Was das Thema mit der Z-Achse betrifft und LightBurn ist es nicht unbedingt der Fehler bei MakerBase...
Für den DLC32 V1 ist alles vorhanden was Du brauchts...gekniffen musst Du leben mit dem was man Dir hier hinwirft.
Der DLC32 ist für einen Diodenlaser und zwei Achsen optimal weil er macht was er soll und es braucht nicht mehr.

Ich kau da gerade auch böse drauf rum aber habe noch diverse Controller die sehr viel besser sind aber wenn es funktioniert will man sich nicht mit noch mehr Kram beschäftigen.

Markus-K40, 2022-09-01:
vielen Dank. Hier sind nun alle Daten:
Board: MKS DLC32 V1.1 (Aufdruck auf dem Board V1.1 - laut Anzeige auf dem Display V1.0)
Display: MKS TS35-R V2.0
Firmware: V2.08(8M.H35.20220105)

Die Steuerung über Lightburn sollte möglich sein, aber klappt derzeit noch nicht. Ich wollte daher die Z-Achse direkt über das MKS-Board steuern. Die Voraussetzung wäre hierfür nur die entsprechenden Button auf dem Display zu haben. Ich habe gelesen, dass es in der finalen Firmware die Z-Steuerung geben soll und die wollte ich installieren und hoffe darauf einen Tipp zu bekommen wie ich die Firmware runter oder aid andere drüber bekomme.

Plottmania, 2022-09-02:
LightBurn ist da ein bisschen grausam... : )
Du hast die Z-Achse aktiviert in LightBurn?

Die MKS Firmware bekommst Du einfach mit dem MKS Tool auf den Controller.
Ich glaub Du bist bei mir im Forum, dann schrei da was sehr viel einfacher ist.

Markus-K40, 2022-09-03:
ja, die Z-Achse ist aktiviert.
Du glaubst richtig - ich schreibe in Deinem Forum und dann wird mein Drama mit der Firmware hoffentlich ein Ende finden.
Vielen Dank für die Hilfe.

hans-de, 2022-10-11:
The latest Update Support the z Axis but it has a lot of new issues. A few days ago i Received a Mail from Makerbase that they are fixing all reported issues and will publish a new Update in September.I retured this shit to Amazon.