makerbase-mks/MKS-DLC32 #46: Laser on full power at idle after updating firmware

Sheaseydel, 2021-11-20:
Updated to newest normal laser firmware and now my laser is on 100% when I power on the dlc32 board?

Sheaseydel, 2021-11-20:
Tested it using the stock dlc_cfg.txt download.

Sheaseydel, 2021-11-20:
Have 3 wire laser module. Hooked up + and - to 12v/24v (pass-through) as it wouldnt work plugged into laser +/-  with original firmware, and signal was connected to S on the ttl plug. It worked fine until I updated firmware, it now will power up using the laser +/- but as soon as the Signal wire is connected the laser powers on full

GadgetX, 2021-11-20:
Unfortunately they change PWM pin on board V2 and firmware works only with V2. I got this some problem 🙄🙄🙄.Looks like they stop supporting board V1.1 😅😅😅

Sheaseydel, 2021-11-20:
I don't see a link to the old firmware yet I only got this board a couple months ago. Does someone have a link to the old firmware for v1.1 board?

21mac, 2021-11-20:
> I don't see a link to the old firmware yet I only got this board a couple months ago. Does someone have a link to the old firmware for v1.1 board?

https://github.com/makerbase-mks/MKS-DLC32/blob/c2451c41e7e60f6465e2da6f59000ffabc4d343f/firmware/Laser/Normal/DLC32_Normal_V1.2.001_8M_H35_0911.bin

Sheaseydel, 2021-11-21:
Thanks so much it is working again!

21mac, 2021-11-21:
> Thanks so much it is working again!

you're welcome.
please don't forget to close this issue :)

closed 2021-11-21 by Sheaseydel

Anderson8153, 2021-11-25:
@21mac , The laser fan is always on on the DLC 32 V2.0! Is it possible to turn off the fan while not using the laser?