makerbase-mks/MKS-DLC32 #47: MKS DCL32 V2.0 XY HOMING

Mihalch999, 2021-11-20:
Hello! I want to customize the firmware configuration on me DLC32. Since there is no search for the home position. How to do it??

21mac, 2021-11-21:
https://github.com/makerbase-mks/MKS-DLC32/blob/main/README.md#parameters-configuration
why don't you take a look at the documentation?
there is the command to enable it and the requirements.

Mihalch999, 2021-11-21:
> команда

There nothing helps.... I need to do next Also, since you are not using a z-axis, you will have to modify config.h and re-upload grbl to your arduino. About line 75 of the config.h you will see these two lines:

#define HOMING_CYCLE_0 (1<<Z_AXIS)
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))

You will need to change them to get homing to work without a z-axis. I'm thinking replacing the two lines with this one should work:

#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))

Mihalch999, 2021-11-21:
But i need to change file config. Where can i take it? I have only DLC32_V2.0_D2.00.3_H35_公版_1108_N.bin

21mac, 2021-11-21:
the firmware is not open source (yet).
you cannot modify it.

you have to change the configuration via GCode + ESP-Codes.
open any software that is capable of sending GCodes manually (like laserGRBL) and send the commands from this documentation:
https://github.com/bdring/Grbl_Esp32/wiki/Settings
like
$Homing/Cycle0=X
$Homing/Cycle1=Y

Plottmania, 2021-11-21:
What a nonsense!
This is set in the GRBL without changing the firmware.
Laser mode or XY mode and would be the GRBL 1.1h
Now enter in the lines:

$3= ?  <----here
$22=?  <----here
$23=?  <----here

Where please is the problem but yes...changing firmware is important for now?!
No, it's all there and just need to understand.

Mihalch999, 2021-11-21:
Thank you, we managed to solve the problem of finding a house. But now a new problem. Something happened with 5 volts. The board no longer has 5 volts, the computer does not see it. Only the VIN indicator is on on the board

Plottmania, 2021-11-21:
Hello,
unfortunately I can't help you, but this is how it looks for me:
A Sculpfun S6 - Laser with one motor per axis.
Limit switches are inductive PL08-N (4-30 volts) and they work very well!
It is a test setup because I always change and try.
My DLC32 V1.0 and the normal firmware (factory settings).
I have already shot me some drivers but the DLC has never bothered all this : )
Never plug anything when the controller is under power...very important.

[image]
[image]

makerbase-mks, 2021-11-25:
> Thank you, we managed to solve the problem of finding a house. But now a new problem. Something happened with 5 volts. The board no longer has 5 volts, the computer does not see it. Only the VIN indicator is on on the board

So how is the problem now? It can work at the first, but now cannot work?

Mihalch999, 2021-11-25:
> > Thank you, we managed to solve the problem of finding a house. But now a new problem. Something happened with 5 volts. The board no longer has 5 volts, the computer does not see it. Only the VIN indicator is on on the board
> 
> So how is the problem now? It can work at the first, but now cannot work?

Good afternoon! Thanks, the problem is solved. The LED on the board has burned out. I replaced it, and now everything works)

closed 2021-11-29 by makerbase-mks

makerbase-mks, 2022-11-15:
Code open source late but finally here: https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE