makerbase-mks/MKS-DLC32 #48: need help wiring co2 laser with dlc32

aroef, 2021-11-23:
I tried wiring the co2 laser with the dlc32 board, but the laser doesn't turn on.
I read on another forum laser turn on with invert the pwm by using a #define INVERT_SPINDLE_PWM On statement in the machine.h with GRBL ESP32

do you have firmware for co2 laser for dlc32 v1.1?

makerbase-mks, 2021-11-24:
@aroef  Did you use the "SPINDLE" interface to connect to the laser? And what is the voltage/current of your laser need?

Plottmania, 2021-11-24:
Perhaps you should ask quite differently....Power supply?
Limit switch and various other things?
A laser and its laser tube or the power supply unit which is to switch...what is connected there and how?
Has everything not really to do with a spindle but would be for me $32=1
The question depending on whether it is 255 or 1000 and also quickly asked what the GRBL looks like?
Here alone the spirits divide: )
I was just wondering if anyone here is of the opinion so a DLC and a few cables make a CO2 laser and that works?
Sorry but if there is a power supply dangling and on this power supply to switch power I ask myself seriously if this is all meant seriously.

There is electricity hanging on which can cause quite nasty injuries!
You really have a clue about what you're doing?

No, I do not think so but I do not care whose hair is sizzling!

PunkRa1n, 2021-11-26:
Hi, maybe you need to update the new firmware, I've had the same issue, a new firmware was released 3 days ago
https://raw.githubusercontent.com/makerbase-mks/MKS-DLC32/main/firmware/TS35/CNC/Normal/DLC32_CNC_V1.2.001_Normal_8M_H35_1009.bin

you will see on the bord the "Laser" led will turn on if you "fire".

But if you measure the voltage of +LASER- will be ca. 24V and idk if it will kill the MYJG-40 transformer
TTL goes to 5V. but I think you still need original regulator because the transformer works with <5V not 24V
But [this guy](https://www.reddit.com/r/ChineseLaserCutters/comments/geym9o/k40_smoothieware_upgrade_mks_sbase_13_ribbon/) connected it directly to a MKS SBASE.
I don't own the sbase and idk what is the output or operation range.

Isn't it the DLC32 for a 24V LED Laser?

A friend told me I should buy the new bord, it was new released. So I've also a new bord, destroyed the LIHUIYU M2 (original) controller and can't use it now...

makerbase-mks, 2021-12-01:
@PunkRa1n Hello, I don't know whether I can get your information correctly. The DLC32 can drive 24v laser of cause, in this situation, you should use the DC24V for power supply.
TTL is 5v pwm output.

I still don't know which point you mentioned has problem. Hope you can feedback.

tschensie, 2021-12-13:
The co2 Laser needs a GND signal to switch on the Laser an 0-5V to regulate the Power (5v = 100%).
This means we need a Pin which switches to gnd at M3 Command and an 0-5v linear output for laserpower (S=1 to S=100 with $30=100 and $31=1)
Or a 5v / 12v / 24v PWM output from 100HZ to 5KHZ to connect something like this to convert the PWM signal to analog voltage: https://www.amazon.de/dp/B0787VZM9B
(10KHZ means 10v output, so pwn should be limited to 5KHZ to get max 5v output from the module)

PunkRa1n, 2021-12-18:
> @aroef  Did you use the "SPINDLE" interface to connect to the laser? And what is the voltage/current of your laser need?

The voltage works between 0.3V (1%) ...5V (100%).
The MYJG-40 transformer can't handle more as @tschensie mentioned.

Will we receive a custom binary for the CO2 laser ?

RandomHacks-Git, 2021-12-18:
I'm using the board with a k40, I just used a small NPN transistor connected to the TTL output with a 1k base resistor. In theory you could use the spindle output which already has a mosfet but for some reason it outputs close to 24v when it is floating (I'm using the 24v from the k40 PSU to power the board) so I just did it this way to be safe. I guess you could use an optocoupler to be extra safe.

The enable pin on the power supply is active low (turns on when grounded) and doesn't need an analog voltage to regulate the power, PWM works just fine.

[image]

aroef, 2021-12-18:
sorry for the late reply
my dlc32 runs fine after updating the firmware with dlc32 v1.1 firmware

dlc32 board power source must be from 24v co2 power supply, connect G to L power supply, and S to in power supply

aroef, 2021-12-18:
[image]
[image]

PunkRa1n, 2021-12-20:
@aroef It works fine with TTL G to L and S to IN
and how did you fixed it with homing? You didn't use the homing function, right? Bc I receive every time ALARM 9 and i see in the console the Z axis goes in one direction and fails. Im also not able to disable the Z axis somewhere.

  
I've added 2 endstops for better homing. The optic entstop didn't work fine. But without Z axis and Z endstop is worthless.
Actually I didn't spent much time to build a Z Bed.
[image]

RandomHacks-Git, 2021-12-20:
I did replace my k40's photo interrupters with micro switches as well. I  just connected the common (COM) of the micro switches to GND and the normally open (NO) to S and it works fine (I don't remember if I had to change any setting apart from $21). I did not build the Z axis yet, it homes fine without it.

[image]
[image]

3Dangel, 2022-01-13:
> @aroef It works fine with TTL G to L and S to IN and how did you fixed it with homing? You didn't use the homing function, right? Bc I receive every time ALARM 9 and i see in the console the Z axis [...]

hello there
would you be able to post some cloeup pictures that shows your connection to the Power supply? 
did you disconnect the digital panel's cable from the In?
and which SW are you using? 

Thank you for your feedback

PunkRa1n, 2022-01-14:
Hi @aroef 
[image]

A better view of my wiring
It works great
$h don't work if you don't have a Z axis. But you can bypass it with $hy and $hx 

Here also my config

```

$0=10 (Step pulse time)
$1=255 (Step idle delay)
$2=1 (Step pulse invert)
$3=1 (Step direction invert)
$4=0 (Invert step enable pin)
$5=1 (Invert limit pins)
$6=0 (Invert probe pin)
$10=3 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=1 (Soft limits enable)
$21=0 (Hard limits enable)
$22=1 (Homing cycle enable)
$23=3 (Homing direction invert)
$24=300.000 (Homing locate feed rate)
$25=1000.000 (Homing search seek rate)
$26=5.000 (Homing switch debounce delay)
$27=5.000 (Homing switch pull-off distance)
$28=8000.000 ()
$30=10000.000 (Maximum spindle speed)
$31=0.000 (Minimum spindle speed)
$32=0 (Laser-mode enable)
$38=0 ()
$40=0 ()
$100=160.000 (X-axis travel resolution)
$101=160.000 (Y-axis travel resolution)
$102=160.000 (Z-axis travel resolution)
$103=100.000 ()
$104=100.000 ()
$105=100.000 ()
$110=6000.000 (X-axis maximum rate)
$111=6000.000 (Y-axis maximum rate)
$112=6000.000 (Z-axis maximum rate)
$113=1000.000 ()
$114=1000.000 ()
$115=1000.000 ()
$120=1000.000 (X-axis acceleration)
$121=1000.000 (Y-axis acceleration)
$122=1000.000 (Z-axis acceleration)
$123=200.000 ()
$124=200.000 ()
$125=200.000 ()
$130=285.000 (X-axis maximum travel)
$131=272.000 (Y-axis maximum travel)
$132=80.000 (Z-axis maximum travel)
$133=300.000 ()
$134=300.000 ()
$135=300.000 ()

```

3Dangel, 2022-01-15:
> Hi @aroef [image]
> 
> A better view of my wiring It works great $h don't work if you don't have a Z axis. But you can bypass it with $hy and $hx
> 
> Actually I do not use the original panel. You can adjust the power with LightBrurn pretty well
> 
> Here also my config Needs also some fine tuning
> 
> $1 is 255 because the motor should lock and don't move around.
> 
> ```
> 
> $0=10 (Step pulse time)
> $1=255 (Step idle delay)
> $2=1 (Step pulse invert)
> $3=1 (Step direction invert)
> $4=0 (Invert step enable pin)
> $5=1 (Invert limit pins)
> $6=0 (Invert probe pin)
> $10=3 (Status report options)
> $11=0.010 (Junction deviation)
> $12=0.002 (Arc tolerance)
> $13=0 (Report in inches)
> $20=1 (Soft limits enable)
> $21=0 (Hard limits enable)
> $22=1 (Homing cycle enable)
> $23=3 (Homing direction invert)
> $24=300.000 (Homing locate feed rate)
> $25=1000.000 (Homing search seek rate)
> $26=5.000 (Homing switch debounce delay)
> $27=5.000 (Homing switch pull-off distance)
> $28=8000.000 ()
> $30=10000.000 (Maximum spindle speed)
> $31=0.000 (Minimum spindle speed)
> $32=0 (Laser-mode enable)
> $38=0 ()
> $40=0 ()
> $100=160.000 (X-axis travel resolution)
> $101=160.000 (Y-axis travel resolution)
> $102=160.000 (Z-axis travel resolution)
> $103=100.000 ()
> $104=100.000 ()
> $105=100.000 ()
> $110=6000.000 (X-axis maximum rate)
> $111=6000.000 (Y-axis maximum rate)
> $112=6000.000 (Z-axis maximum rate)
> $113=1000.000 ()
> $114=1000.000 ()
> $115=1000.000 ()
> $120=1000.000 (X-axis acceleration)
> $121=1000.000 (Y-axis acceleration)
> $122=1000.000 (Z-axis acceleration)
> $123=200.000 ()
> $124=200.000 ()
> $125=200.000 ()
> $130=285.000 (X-axis maximum travel)
> $131=272.000 (Y-axis maximum travel)
> $132=80.000 (Z-axis maximum travel)
> $133=300.000 ()
> $134=300.000 ()
> $135=300.000 ()
> ```

@ PunkRa1n

thank you for posting your info and picture. 
may I ask why didn’t you use $32=1 to enable the laser!
This way it will not enable the Z axis 
As for power adjustment in Lb
Did you go through the testing to see what is the Maximun power you can use for your K40? So it will not damage the laser tube?

thanks agsin

PunkRa1n, 2022-01-15:
Hi @aroef 

Thanks, that was the missing piece of information what I've needed, the laser mode.

I didn't had much time for testing. 

The transformer don't destroy the tube. It can't output more then ~40W if you set the output to 100% like on the 7 segment panel to 99.9. It's the same if you set the TTL to high like 5v
The signal looks like this
[image]
So you can't overpowering it. 
But it works pretty well in a rage of 20-55%. 

But I've found an interesting article about the k40 Lasertube. 
It isn't a 40W tube.

https://k40.se/k40-laser-troubleshooting/dead-lasertube/

As mentioned in the article, it should be installed additionally a 30mA meter.

3Dangel, 2022-01-15:
Hi PunkRa1n

I'm not aroef. :-)
and you are welcome about the $32=1

The consensus is that the laser tube  in our K40,  should not over powered beyond 15mA.
the 7 segment display (the Digital controller panel) is not accurate and by reaching beyond 30, would shorter your Co2 laser tube's life spam.
I suggest you to join FB K40 user group and you will  find much more of information there.
Hope you are not driving your tube beyond 30 (which is the Max power to NOT kill sooner your tube).
> Thanks, that was the missing piece of information what I've needed, the laser mode.
>
> I didn't had much time for testing.
>
> The transformer don't destroy the tube. It can't output more then ~40W if [...]

3Dangel, 2022-01-23:
[image]

3Dangel, 2022-02-21:
here is what I have as wiring:

[image]

3Dangel, 2022-02-24:
> I'm using the board with a k40, I just used a small NPN transistor connected to the TTL output with a 1k base resistor. In theory you could use the spindle output which already has a mosfet but for [...]
hi [RandomHacks-Git]

I'm having a bit of issue with my connection...
my board is a V2.x
it has a dedicated 3 pins TTL as well as the SPINDLE out out of 2 pins
it seems my TTL is not the same as the SPINDLE's pins; the does not go 24V when floating. BUT the spindle's does.

If connected the TTL's Gnd to the L, the laser will trigger on and fire at powering up the unit non stop.
it connecting the Spindle's Gnd to L, it will act correctly with the digital panel which controls the MAX power limit, but I am loosing the PWM  contol ( never got it to work... I'm using an external power supply)
would you be able to share couple more picture of your set up and if time permits, how you connected the NPN transistor with the resistor... I see your picture, but cannot make out where you are connecting the connector to (from - to).

really appreciate your feedback!!! 
Thank you in advance!!

kubicek, 2022-03-02:
I thing the mosfet should be connected like this. I have not tested it yet.
[image] (picture from https://forum.makerforums.info/t/grbl-1-1f-config-with-prontoneer-cnc-shield/9155/12)
.

fr4nc3sco, 2022-09-07:
Hi,
i have a problem getting my 50w laser to work with the mks dlc32 2.1v basically my co2 power supply only has:
TH TL WP G IN 5V
the only links I found working were:
power supply  CO2                  MKS
TL                                           -SPINDLE
G                                            GND

The problem is so set up it delivers at most only 12-13ma against 22 but it delivers with the original m2 nano what am I doing wrong? is there a solution to deliver full power?

maadmackz, 2023-01-14:
can we not just do this like it says???
[image]

capt1701a, 2023-01-26:
@3Dangel, hope you or someone can help as I have the same Makerbase MKS DLC32 V2.1 as commented on Feb 21, 2022  but I have the type 2 power supply for the K40 laser engraver.

[image]

You mention, due to changes in V2.x that TTLs Gnd has changed and it is a REAL GND, causing the K40 to fire at powering ON. The remedy it to bring the Gnd from the Spindle and it will wok. The diagram does not show going to the negative on the spindle. I'd like clarification whether wiring either way causes any issue? 

Also my laser head does not register it is home when the laser head hits the limits and keeps banging against them $32 is enabled. These wires for the limit switches are in the G and S on the motherboard. I get the following error on lightburn when the head hits the limit
  
[image]

This is the details from Lightburn

$0=10
$1=200
$2=0
$3=1
$4=0
$5=0
$6=0
$10=0
$11=0.010
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=3
$24=75.000
$25=2400.000
$26=50.000
$27=4.000
$28=1000.000
$30=1000.000
$31=0.000
$32=1
$38=1
$37=1
$40=1
$100=160.000
$101=160.000
$102=160.000
$103=100.000
$104=100.000
$105=100.000
$110=500.000
$111=5000.000
$112=1000.000
$113=1000.000
$114=1000.000
$115=1000.000
$120=800.000
$121=300.000
$122=1500.000
$123=200.000
$124=200.000
$125=200.000
$130=300.000
$131=200.000
$132=50.000
$133=300.000
$134=300.000
$135=300.000

Any help would be appreciated

3Dangel, 2023-01-26:
Yup going to G of the spindle
Only one line connected from spindle’G to the power supply of the laser.
Limit switch
You would need to try on error.  Multimeter to test first if it is working
or not
Then connected onto DLC32, does it stops the motor manually triggering it?;
I uploaded my config in k40 FB group..
My limit switch x= left end, y=left back

DLC32 needs the ground from the main power supply

Joint FB group: Chinese K40 laser group
File & posts are in there
Join LB forum more info in there
Key word = DLC32

Good luck

Ilin32, 2023-08-04:
Здравствуйте. Сталкнулся с аналогичной проблемой. 
Решилась следующим путем.
Подключение: S=>IN
                            TTl=>G
Прошивка под лазер: v2.30_H35_2022_1223_N_Z_X_001.bin

[image]
[image]
[image]

matonhodze, 2023-09-08:
solder laser PSU (N) to the left leg of the mosfet on board 
then connect laser PSU (L) to pin SCL on board  and you are done you wont need a potentiometer .thus arrangement can work with any laser software   
[image]

Flipcnc, 2023-10-13:
Hi friends,First time on Github,not to sure how it works,but i have a few questions.i have the same setup as Llin32 above.i just want to know if his setup and wiring is ok and if it is working.i dont know how to ask him directly.

Flipcnc, 2023-10-13:
[image]
[image]
[image]

maadmackz, 2023-10-13:
> Hi friends,First time on Github,not to sure how it works,but i have a few questions.i have the same setup as Llin32 above.i just want to know if his setup and wiring is ok and if it is working.i dont know how to ask him directly.

IS YOUR BOARD A V2.1??? They are slightly different.

Flipcnc, 2023-10-16:
so do i use GND by TTL section or - by Spindle section to conect to G terminal on laser power suply?

Flipcnc, 2023-10-16:
[image]

matonhodze, 2023-10-16:
Solder (N) to left leg of mosfet as in picture then ( L)  connect it to SCL
pin you are done this should work

Rginoris, 2025-02-01:
Please guys tell me how I have to connect this, I have already tried several ways without any result.

[image]

matonhodze, 2025-02-02:
https://github.com/makerbase-mks/MKS-DLC32/issues/493

matonhodze, 2025-02-02:
Try this however it fails then let me know and will give u the one i used
for my laser that involves soldering the (IN) wire to mosfet

BOOSTED1SS, 2025-03-12:
Instead of soldering to the mosfet could I use one of these[image]

matonhodze, 2025-03-13:
> Instead of soldering to the mosfet could I use one of these5740ED58-7C85-4324-AF83-5997940E71F3.png

The issue is u need  an out put that is no more than. 5v that is withn
range of the power supply (IN) voltage hence if that coupler is using a
relay its going to be tricky as u need an analogue signal to vary u laser
power (0-5v) range

LesterManders, 2025-03-28:
> [image] [image] [image]

When I do this. The laser starts shooting when the control board loses power. How to prevent this?

Ilin32, 2025-10-24:
> > [image] ![лазер2](https://private-user-images.githubusercontent.com/147820401/274884609-b04c071a-6fff-4e67-8753-a2d7646c7641.jpg?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjEyNjUwOTQsIm5iZiI6MTc2MTI2NDc5NCwicGF0aCI6Ii8xNDc4MjA0MDEvMjc0ODg0NjA5LWIwNGMwNzFhLTZmZmYtNGU2Ny04NzUzLWEyZDc2NDZjNzY0MS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMDI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTAyNFQwMDEzMTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xZTNlYjhkZmVkMjA2NzFkOTdhMDU3M2U2OTUyMTFjOTlmZjVlYmM1NTRlZWQ3Y2EyYzcxZDdkOGJiMDAwNWU0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.PDAGyqahJS5No2HS4CDfY0q3gMZ98yL3UzGIB-QwKT0) ![лазер1](https://private-user-images.githubusercontent.com/147820401/274884628-74ac647d-dd18-42e1-8e76-87045663d73d.jpg?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjEyNjUwOTQsIm5iZiI6MTc2MTI2NDc5NCwicGF0aCI6Ii8xNDc4MjA0MDEvMjc0ODg0NjI4LTc0YWM2NDdkLWRkMTgtNDJlMS04ZTc2LTg3MDQ1NjYzZDczZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUxMDI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MTAyNFQwMDEzMTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMDEwNzcyYjI1MDg5YjY5ODM1Njg1NDE0YjBiY2ZkMTM2MTg1MTUxOGU3ZDhiYWM1OWRiYWVhYzk2NGI5MzY4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Q44_lHyDxVoh2COgMucxom9HunMff3wQVGpj9AQqRkU)
> 
> Когда я это делаю. Лазер начинает стрелять, когда плата управления обесточивается. Как этого избежать?

Для этого кнопка включения между G и L