makerbase-mks/MKS-DLC32 #51: where does the "For CNC, CoreXY machine(For beta)” firmwire can be find?

lymanzhang, 2021-11-27:
I need to download the firmwire: **For CNC, CoreXY machine(For beta)**, I just click on the link: 
https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/CNC/CoreXY, 
but the link is not exist! So where it can be find?

trashypopashy, 2021-11-27:
You will need to look at the history-folder of this page.

trashypopashy, 2021-11-30:
@mks-viva @solawc @makerbase-mks @MS1987 I am still waiting for an updatet firmware for a CNC (normal, Board V1).

makerbase-mks, 2021-12-01:
@trashypopashy Actually, there is a version has been updated 8 days ago: https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/CNC/Normal. Can't it meet your requirement?

@lymanzhang We have found the version for CoreXY has some bugs and have fixed, but still wait for test and confirm, then will be upload.

trashypopashy, 2021-12-01:
> @trashypopashy Actually, there is a version has been updated 8 days ago: https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/CNC/Normal. Can't it meet your requirement?
>
> @lymanzhang [...]

@makerbase-mks filename an size (every single byte) is exactly the same as the version, that is some  months older. So i am sure, this is not an upfated version.

trashypopashy, 2021-12-05:
@mks-viva @makerbase-mks @solawc @MS1987 please give me an updated firmware.

lymanzhang, 2022-02-23:
@[trashypopashy When will the new firmware supporting core XY CNC be released?

trashypopashy, 2022-02-23:
I am just a user. Can't help You, sorry.