makerbase-mks/MKS-DLC32 #53: Air assist pin

FrancescoIT, 2021-11-28:
I use lightburn, to which pin should I connect the air assist?

makerbase-mks, 2021-12-01:
I don't think there is such function pin on dlc32, can you tell me how the air assist be drive?

FrancescoIT, 2021-12-01:
[image]

if the automatic air assist option is activated, lightnurn enters the M8 command in the gcode the pin becomes high (the air assist turns on) with M9 the pin goes low (it turns off).
This function is very important for laser machines, because if the air is not activated during cutting, the material catches fire.

this is an example of how the pin is configured on smoothiware, I hope it can help you to insert this indispensable function if you want to make your product more professional.
## Switch module for air assist
switch.fan.enable                            true
switch.fan.input_on_command                  M8             #
switch.fan.input_off_command                 M9             #
switch.fan.output_pin                        2.7              #
switch.fan.output_type                       pwm              # pwm output settable with S parameter in the input_on_comand
#switch.fan.max_pwm                          255              # set max pwm for the pin default is 255

szogya, 2021-12-02:
I think the air assist should be a must have function for a laser machine at least for safety reasons.
While I don't see pin for the air assist, there should be a firmware version, source code or a function where we could use another pin remapped for this. The ESP controller should have the capability for doing that.

FrancescoIT, 2021-12-02:
@makerbase-mks do you plan to add this option in the next firmware release?

RandomHacks-Git, 2021-12-02:
I'm also waiting for this.

TeamZulu, 2021-12-05:
waiting too. this funktion is missing and you need it!!

FrancescoIT, 2021-12-05:
strange that none of the team responds, you think we are not able to do it or it is just listlessness ?

TeamZulu, 2021-12-05:
I have no idea. but if there is no solution, i buy another board that can air assist. i currently have my air assist manually on a counter. I would prefer an automatic solution via the board

trashypopashy, 2021-12-06:
The DLC32 is claimed to be for use in laser engraving machines, but has no airassist pin. Maybe one can use an endstop-signal pin.

Once again @makerbase-mks @solawc @mks-viva @MS1987 go open soure if you will stay unacceptable bad supporters.

makerbase-mks, 2021-12-06:
Sorry for delay response. 
So the air assist just need a on/off pin, we will add it as soon as possible.

FrancescoIT, 2021-12-06:
> Sorry for delay response.
> So the air assist just need a on/off pin, we will add it as soon as possible.

yes, only on/off pin ( ON on gcode command M8 and OFF on M9)

thanks, please hurry, I bought 3 boards and I need to assemble them ;-)

makerbase-mks, 2021-12-06:
@FrancescoIT I still have a question, does it need to use 12/24V to drive? As there are only TTL level on dlc32 board, if need the 12/24V, should add an extern mosfet .

FrancescoIT, 2021-12-06:
> @FrancescoIT I still have a question, does it need to use 12/24V to drive? As there are only TTL level on dlc32 board, if need the 12/24V, should add an extern mosfet .

it is fine a ttl logic level for the mosfet.  Thank you

TeamZulu, 2021-12-06:
I do not use the 12 / 24v connection.  I have a mosfet with an emergency stop for the laser module and ttl to control the laser module

makerbase-mks, 2021-12-06:
@TeamZulu @FrancescoIT Thanks for feedback.

FrancescoIT, 2021-12-06:
> I do not use the 12 / 24v connection.  I have a mosfet with an emergency stop for the laser module and ttl to control the laser module

we are talking about air asstit, I do not see the relevance with what you say

TeamZulu, 2021-12-06:
@FrancescoIT Yes, I am assuming.  I suspect the question from makerbase because of the output is whether this is used.  Maybe you want to pin this over as an air assist exit

speedymk1, 2021-12-08:
Hello, I also need one pin for the air assist. To use an external relais or something like this is no problem for activate the air compressor. Give us a solution to use it.

FrancescoIT, 2021-12-10:
@solawc Hi, sorry for the inconvenience, I saw that there was an update of the fw, have you implemented the air assist?

speedymk1, 2021-12-10:
> @solawc Hi, sorry for the inconvenience, I saw that there was an update of the fw, have you implemented the air assist?

Solawc isn't in this conversation

medline2c, 2021-12-10:
Je suis également très intéressé par cette possibilité ! je rejoints @szogya : c'est indispensable et ne devrait pas être une option... un niveau TTL permet de contrôler un relais qui allumerait et eteindrait l'assistance aérienne...

TeamZulu, 2021-12-10:
Air assist on 12/24 pin or spindle motor pin, That would be a great solution

TeamZulu, 2021-12-10:
Simple on/off Funktion for relay

RandomHacks-Git, 2021-12-10:
Not the spindle pin as it is the same as the TTL pin (IO32) and the 12/24 connector is just a power output with no control.
I think the best pin would probably be IO22 (marked probe on the board).

speedymk1, 2021-12-10:
I think the guys of makerbase-mks will not implement this function.
I will buy an other board...

TeamZulu, 2021-12-10:
@speedymk1 i think too

TeamZulu, 2021-12-10:
Back to marlin laser

makerbase-mks, 2021-12-13:
New firmware has released: Join the function of MISP, you can use M7/M9 to control
https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/Laser/Normal

RandomHacks-Git, 2021-12-13:
> New firmware has released: Join the function of MISP, you can use M7/M9 to control(using SCL pin) https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/Laser/Normal

Will this update be available for the 2.4" LCD version?

FrancescoIT, 2021-12-13:
have updated the fw, i have already tested and it works fine #73

solawc, 2021-12-14:
Currently, the firmware with 3.5-inch screen and the firmware with 2.4-inch screen have added the switch function of M7/M9, and the M8/M9 is used for control on the CNC. @FrancescoIT Of course I have a question, will M8 and M7 be used at the same time? Do they have to use two different pins? When connecting to this device, if the pin level itself does not change, will the motherboard pin signal be pulled down?

solawc, 2021-12-14:
Can you use your device to test it for me? Connect your air-assisted interface to the SDA pin, and power on again to see if the machine can start normally. @FrancescoIT

makerbase-mks, 2021-12-14:
> > New firmware has released: Join the function of MISP, you can use M7/M9 to control(using SCL pin) https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS35/Laser/Normal
> 
> Will this update be available for the 2.4" LCD version?

Have updated: https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/TS24

closed 2021-12-15 by makerbase-mks

FrancescoIT, 2021-12-15:
> Can you use your device to test it for me? Connect your air-assisted interface to the SDA pin, and power on again to see if the machine can start normally. @FrancescoIT

I try and let you know

killrob, 2022-01-22:
which is the SCL pin on MKS DLC32 V2.1? please

3Dangel, 2022-01-23:
> which is the SCL pin on MKS DLC32 V2.1? please

it is the bare pins that are located in front of the PROBE.
use the SCL and G to drive a relay. I tested on my SSR dc-AC and it is working. 
Lightburn : you need to enable the M7 in device setting.

hope this helps

makerbase-mks, 2022-01-23:
> which is the SCL pin on MKS DLC32 V2.1? please

 @killrob  Here is the relavant wiki: https://github.com/makerbase-mks/MKS-DLC32/wiki/Hardware-Wiring#supplementary-note

killrob, 2022-01-23:
ok i've tryed and it works as intended, thank you

HunLucas, 2022-05-10:
Greetings!
SCL pin have 5v output or GND? Can I drive an 5v relay with optocoupler to protect the board?

killrob, 2022-05-10:
SCL is only a signal pin, you can connect to 5V and G of the Z switch endstop

iron1800, 2022-11-28:
Anybody know what the 3v3 pin in on the 4 pin connector?  My laser head has crosshairs that use 3v3 pins can this 3v3 and a gnd trigger the crosshairs from this 3v3 pin and is there anyway to program grbl to just activate this pin say for framing?

oldschool91, 2023-06-15:
hello my laser engraver (T-Trees TTS55 10W) is equipped with an MKS LTS v1.1 card, there is an SCL output, so I made the connections as on an MKS DLC32, but nothing happens at the output SCL is always supplied with 3v whether I activate M7 or M8 in lightburn.
What should I do so that the SCL output cuts off during burning or the opposite? Must write in ligtburn a start and an end gcode? if yes which?
Does the firmware of this card allow it?
Thank you for your help

iron1800, 2023-06-15:
Sorry but I'm not familiar with either the card or the laser or I would help you out.  I have the mks dlc32 and the xtool d1 laser.  The dlc 32 has a 12v power output that goes to the laser with ground then there is the ttl.  From what I have gathered about the ttl is is what varies the output of the voltage to the laser as well as turning it on and off.

andreasr81, 2023-07-12:
> [image] hello my laser engraver (T-Trees TTS55 10W) is equipped with an MKS LTS v1.1 card, there is an SCL output, so I made the connections as on an MKS DLC32, but nothing happens at the output SCL [...]

Hi i have the same board in my ray5 10w version, did you have any solution? i lealy want a air assist control and dont want to buy a new board.
Maybe the dlc32 firmware will work?

regards

bouba830, 2023-09-24:
Bonjour , sur une MKS dlc V2.0 avec tft 35 v.1 , j'imagine qu'il n'y a pas d'interface 12c?

Technolitix, 2023-12-11:
Why, do you not implement the I2C bus when you already have it on the board, to add a eg. I2C-relay board or a mcp23017 expansion board? This would increase the I/O of the board.

3Dangel, 2023-12-11:
The why would be to ask to. GRBL FW development team.
Makerbase does not develop GRBL.

linewallker, 2024-07-22:
Do you know that you can use random ENABLE pin signal for automatic air assist without any commands in gcode ? The relay will turn on every time when motors will move, you can add switch on signal cable to the relay to control if you want air assist to be working or not. I am using this method since usage of Arduno cnc shield and it works as charm ;)