makerbase-mks/MKS-DLC32 #61: MKSLaser app

Charus-OV, 2021-12-02:
Hello. Add a font color to the app. I use a dark theme in my phone and fonts are not visible. If you use a light theme then everything is fine
[image]
[image]

una2012, 2021-12-06:
Can anyone help me how to connect the phone to the MKSDLC 2 board
`Waiting for connection...
ok
[VER:1.1.2021051202:]
[OPT:PHSW]
[MSG:Using machine:MKS DLC32]
[MSG:Mode=STA:SSID=Dejan:Status=Connected:IP=192.168.1.10:MAC=94-3C-C6-67-1D-2C]
ok
Starting stream
WebUI Disconnected!
[MSG:WiFi Off]
[MSG:Client Started]
[MSG:Connecting Dejan]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connected with 192.168.1.10]
[MSG:Start mDNS with hostname:http://grblesp.local/]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
Stream completed in 0:06
$S
$Sta/SSID=Dejan
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=MKS_DLC
$AP/Password=******
$AP/IP=192.168.4.1
$AP/Channel=1
$System/Hostname=grblesp
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=STA
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=XY
$Homing/Cycle1=
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=10
$Stepper/IdleTime=25
$Stepper/StepInvert=
$Stepper/DirInvert=X
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=Off
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=Off
$Homing/DirInvert=X
$Homing/Squared=
$Homing/Feed=300.000
$Homing/Seek=1000.000
$Homing/Debounce=250.000
$Homing/Pulloff=1.000
$Spindle/PWM/Frequency=1000.000
$GCode/MaxS=1000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=On
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=1000.000
$Spindle/Type=LASER
$flame_status=Off
$gyro_status=Off
$beep_status=Off
$Language=0
$rollerMode=On
$X/StepsPerMm=80.000
$Y/StepsPerMm=80.000
$Z/StepsPerMm=80.000
$A/StepsPerMm=100.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=6000.000
$Y/MaxRate=6000.000
$Z/MaxRate=6000.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=500.000
$Y/Acceleration=500.000
$Z/Acceleration=500.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=285.000
$Y/MaxTravel=272.000
$Z/MaxTravel=80.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=0.250
$Y/Current/Run=0.250
$Z/Current/Run=0.250
$A/Current/Run=0.250
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=0.125
$Y/Current/Hold=0.125
$Z/Current/Hold=0.125
$A/Current/Hold=0.125
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=16
$Y/Microsteps=16
$Z/Microsteps=16
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
ok
WebUI connected!`

My $S file

makerbase-mks, 2021-12-14:
Have you fixed it?

una2012, 2021-12-14:
No, I still can't connect to my Android devices.

makerbase-mks, 2021-12-15:
@Charus-OV , Of cause, I have asked the app developer to fix it.

makerbase-mks, 2022-01-07:
Have fixed on the latest version on google play.

closed 2022-01-07 by makerbase-mks

Txuxto, 2022-01-22:
Eu instalei a placa MKS DLC 32 v2.0 in 3018 cnc, porém ao ligar o spindle a mesma indica erro e perde a comunicação.
O que pode ser?