makerbase-mks/MKS-DLC32 #63: Config File for FluidNC

diruuu, 2021-12-02:
I successfully compiled [FluidNC](https://github.com/bdring/FluidNC) into this board. But I can't find any config example for FluidNC to have a working GRBL running on this board. Has anyone tried it? Can someone share the config file for it?

FrancescoIT, 2021-12-02:
I also installed fluid nc, but in the example configuration of the laser there are so many parameters, even using the config.txt of gbrl 1.1 which they say is compatible

szogya, 2021-12-03:
I would be interested to see this also. FluidNC seems to be a good alternative and more flexible.

diruuu, 2021-12-03:
After experimenting with the docs, and also with help of the author of FluidNC, I came up with [this working config file](https://github.com/diruuu/FluidNC/blob/main/example_configs/MKS-DLC32-v2.0.yaml). This works for MKS-DLC32 v2.0. Please be note that V.1.x has different pin number for probe so if you have the board, please make sure to check the schematics.

solawc, 2021-12-07:
> I successfully compiled [FluidNC](https://github.com/bdring/FluidNC) into this board. But I can't find any config example for FluidNC to have a working GRBL running on this board. Has anyone tried it? Can someone share the config file for it?

When using FluidNC, it uploads the configuration through the yaml file, so you should first ensure that the yaml has been uploaded successfully and read correctly, and the rest is to ensure that the GPIO configuration is normal, the I2S mode and the output mode

bnorth12, 2021-12-17:
What PINs are the axis limit switches linked to on the board?

Cncninja, 2021-12-31:
Why run fluidNC the base software is grbl 1.1 same as fluid (Core motion) you can use any sender( CNCjs,io,bcnc,etc) via usb or the Web UI and the LCD still works. I don't know of any feature you gain with fluidnc and the opposite you lose the lcd.

maxholgasson, 2022-02-28:
How did you install FluidNC on MKS-DLC32? Does the limit switches, z-axis work? Can you also continously move axis with long press and move single step with short press?

tuxcnc, 2022-05-22:
> Why run fluidNC the base software is grbl 1.1 same as fluid (Core motion) you can use any sender( CNCjs,io,bcnc,etc) via usb or the Web UI and the LCD still works. I don't know of any feature you gain with fluidnc and the opposite you lose the lcd.

Well...
I don't know any reason for the DLC32 firmware source are closed.
The MKS DLC32 is great and low cost hardware,  but unusable for me, because I need bluetooth, not wifi, and I'm not interested in touch screen.
I hope the FluidNC is solution for me.
That is all.

geogarden, 2022-05-22:
> Well... I don't know any reason for the DLC32 firmware source are closed. The MKS DLC32 is great and low cost hardware, but unusable for me, because I need bluetooth, not wifi, and I'm not [...]

Right.
Question for MAKEBASE - Will there be source code and when?

creechoftheeast, 2023-03-21:
I'm struggling to get FluidNC on this board; anyone got some pointers? I'm getting stuck in a registry loop, it seems when running the FulidTerm file from the fluidfs file.