makerbase-mks/MKS-DLC32 #64: ESP32-WEB

Ondrashek, 2021-12-06:
1. Uploading file have no uploading bar - don't know if large file is uploading
2. There is no remove file function - needs to be deleted from sd card
3. I get multiple upload error on different files (no. 5) and ESP32-WEB and panel will restart

solawc, 2021-12-07:
> 1. I get multiple upload error on different files (no. 5) and ESP32-WEB and panel will restart
> 2. Can you separate status window in new tab instead on SD File tab?
> 3. Status window (Laser power, Speed) during engraving/milling have bad translation in English:  速率(%) 100 功率(%)

Thank you for your feedback. What you mean in English should be (rate)% (power)%, right?

Plottmania, 2021-12-07:
Hello, I have the impression one has not really got a grip on it all!
There are versions, different boards and almost no documentation.
Neither one has the possibility something to nor to program itself.
What is meant by: Override
In the English language is down in the menu "Chinese"...so simply it is....looks there times someone after?
It is just cruel what is going on here and MKS must not be surprised if the thing goes to the wall.
For me I can only hope FluidNC works and the WebUI can be customized....at least this part is really OpenSource.
The DLC series is very good but the DLC32 and the different versions and guessing which software on which controller with which version.
Who just wants to buy a controller gets here plenty of problems with a system which is not really ready.
I have only DLC32 V2.1 and no display which is pointless if the WebUI does not work properly.
A DLC32 V2.1 can also be used with LightBurn but then you don't need a DLC32 ... the simple and stress-free DLC 2.0 is enough.

It's just my opinion but "user friendly" I understand quite different.
Sorry for my bad English but this is still better than my Chinese.

Ondrashek, 2021-12-07:
> Hello, I have the impression one has not really got a grip on it all! There are versions, different boards and almost no documentation. Neither one has the possibility something to nor to program [...]

Hi, please don't hate this board. DLC 2.0 is older board. DLC32 is newer board like two months old. I am sure that all will be great as DLC 2.0.

@solawc:  I am not sure if first is rate and second power. I didn't try it. 
[image]

deltazerorsan, 2021-12-07:
> Hi, please don't hate this board. DLC 2.0 is older board. DLC32 is newer board like two months old. I am sure that all will be great as DLC 2.0.
>
> @solawc: I am not sure if first is rate and second [...]

For sure, tested, the left one is feed overide. The second one presumably power override or spindle speed override...

Plottmania, 2021-12-07:
Ondrashek 

[Hi, please don't hate this board. DLC 2.0 is older board. DLC32 is newer board like two months old. I am sure that all will be great as DLC 2.0.]

I like the DLC32 and this has a lot of potential but this is not used and not really maintained.
All my DLC's run well but the DLC32 could run much better if only they wanted to.

It is very simple, invest some time and listen to the wishes that paying customers have.
You can not implement everything but a lot and should start with it.

Documentation, description and very important...describe correctly!
Still very unstructured regarding the updates of the revisions of the controller.
Very cruel still the WebUI and there are already many people help...

Ondrashek, 2021-12-07:
> Ondrashek
>
> [Hi, please don't hate this board. DLC 2.0 is older board. DLC32 is newer board like two months old. I am sure that all will be great as DLC 2.0.]
>
> I like the DLC32 and this has a lot of [...]

Yeah, you are right, but still makerbase response on github issues, facebook group and support email. So it's not like ***censored company*** when they release a new laser model on kickstarter and stay quiet. I see there will be a lot if works on this board and makerbase trying to develop. Will be really helpful when makerbase release firmware source code.

Plottmania, 2021-12-07:
Sorry, who sells to the West should also provide the West with the things that are normal in the West.
What is GitHub, please?
For me OpenSource and what do I have here?
I would love to help but no one has answered my questions yet.
My wishes (I am not alone) were not implemented!
There is no whining and no complaining...only support.
But all no problem if I get FluidNC to run, Web3DUI works and all others have done the work I can at least buy a cheap controller : )

Jawoll....so Business works

Translated with www.DeepL.com/Translator (free version)

solawc, 2022-01-12:
the issue has been fixed

closed 2022-01-12 by solawc