makerbase-mks/MKS-DLC32 #65: Fw version

FrancescoIT, 2021-12-06:
can you tell me why any firmware version I install $ i always tells me the same version ?
[image]

una2012, 2021-12-06:
I have same problems

solawc, 2021-12-07:
> I load the FW V2.0.4_H35_20211123_N.bin and the version on the board always remains the old one, even though the upload was successful [image]

Thank you for your feedback. It is true that the version number of this place has not been modified. This question will be modified today. The version number displayed on the screen should have been modified

solawc, 2021-12-11:
In the new version update, the correct version number can be queried normally through LB, or through $I.

FrancescoIT, 2021-12-11:
sorry if I ask you here, can you tell me when the pin will be implemented to automatically manage the air assist? 
see issue # 53.
thanks again for the support

closed 2021-12-14 by makerbase-mks