makerbase-mks/MKS-DLC32 #66: WIFI password with ! does not work

insensitiveclod, 2021-12-07:
Hello,

I have a wifi-password that ends with the '!' character.
This seems to be a problem for the current firmware.
Entering it via WEBUI fails to work
Using the GCODE interface to set it results in a parsing error for the interpreter (forgot exactly what it returns); it seems to parse it in a special manner and attributes special meaning to it.

When removing the ! character from the end of my (existing) wifi-network password. it started to work.

To reproduce:
- Try entering a wifi-password ending with ! in the wifi-gui
- Use gcode $Sta/Password = blablabla!

I wasnt able to find a way around this using escapes or quoting; and if there is a way to get it to work, the WEBUI also doesnt know about it (yet)

Cheers !

solawc, 2021-12-09:
> 

Hello, is the "!" you used in Chinese or English? Currently, Chinese characters cannot be recognized correctly

solawc, 2021-12-09:
> Hello,
>
> I have a wifi-password that ends with the '!' character. This seems to be a problem for the current firmware. Entering it via WEBUI fails to work Using the GCODE interface to set it results [...]

Hello, is the "!" you used in Chinese or English? Currently, Chinese characters cannot be recognized correctly

ryanm101, 2022-02-02:
I have this issue too, characters are in English

torenich, 2022-02-03:
It will work if you use one of the available touchscreens to set the password. Unfortunately the ! character is handled differently by the console. There _is_ a way to do it, but you have to key it in the console in a very specific manner. I've never done it, but there is info out there showing how to do it.

ryanm101, 2022-02-03:
oh just the ascii code for it.. interesting thank you.. 

I dont have an LCD as the plan was to run this via octoprint or a headless server. 

offtopic: Any recommendation on linux software to manage the cnc machine?

torenich, 2022-02-04:
Octoprint works pretty well with the "Better Grbl Support" plugin, at least the last time I used it a few months ago. There is also a server called LinuxCNC. I've never used it, but it's pretty popular.

insensitiveclod, 2022-02-04:
I've been using https://github.com/vlachoudis/bCNC

There are some issues with using it together with the ESP32-based boards however; the software doesnt quite like the serial output from the board whenever there's a machine reset.
The way that most esp boards are hooked up to USB serial is, however, to perform a reset whenever DTR (i believe) is triggered. And a lot of software triggers the DTR upon startup. 
There's some small edits one can do in bCNC to get it to play well with the Makerbase-DLC32.
bCNC can also push G-Code to a tcp-socket (useful), and has a way to vizualize what the machine will do, as well as basic scripting and manual g-code control.

abinoer, 2022-02-25:
The English symbol "!" It is normal

closed 2022-02-25 by abinoer