makerbase-mks/MKS-DLC32 #67: TS35-R interface bad language + ideas

Ondrashek, 2021-12-07:
Language:
Cancle => Cancel
Postition => Position
Postitioning => Positioning
Cavre => Carve

Ideas:
- Framing from file is very very slow during reading large gcode even when I want to repeat
- When framing is finished next popup is carving but I miss that is carving (no headline of starting carving) so I confirm and full power laser come out to my eyes :D 
- Speed, laser and passes during printing shows 100%, 1pass but 100% of what I set? Better will be show my speed and laser from gcode as F120, S800 which is 120mm/min and 80% power, passes from gcode will be great too 
- Please add framing multiple times and speed rate of framing
- Please please add function to change screen interface icons and custom language (as TFT35)

Charus-OV, 2021-12-07:
> 

To quickly make a detour along the contour. You need to insert the ready-made g-code into the program and run it. This way it will be faster and you can edit the code as you nee
[image]
d.

solawc, 2021-12-09:
> Language: Cancle => Cancel Postition => Position Postitioning => Positioning Cavre => Carve
> 
> Bugs:
> 
> * Framing from file is very very slow during reading large gcode even when I want to repeat
> * When framing is finished next popup is carving but I miss that is carving (no headline of starting carving) so I confirm and full power laser come out to my eyes :D
> * Speed, laser and passes during printing shows 100%, 1pass but 100% of what I set? Better will be show my speed and laser from gcode as F120, S800 which is 120mm/min and 80% power, passes from gcode will be great too
> 
> Ideas:
> 
> * Is possible to add remaining time in progress?
> * Please add framing multiple times and speed rate of framing
> * Please please add function to change screen interface icons and custom language (as TFT35)

Thank you for your feedback and suggestions. The language translation errors will be fixed as soon as possible.

solawc, 2021-12-11:
GRBL is generally very fast to execute nc, and the actual experience of displaying speed/power in real time is also very bad. Although it is possible to directly display the relevant value after parsing an instruction, the refresh on the screen is actually very frequent. Believe that the performance of MCU is limited, it will put a lot of performance on refresh instead of motion control.

solawc, 2022-01-12:
The language error issue has been fixed

closed 2022-01-12 by solawc