makerbase-mks/MKS-DLC32 #71: Unable to find mksLaserTool_V1.X.X software

lgm42, 2021-12-10:
I've just received my MKS DLC32 and I would like to configure Wifi to be able to connec tto it with my phone.
As I've seen in documentation (https://github.com/makerbase-mks/MKS-DLC32/blob/main/doc/DLC32%20Firmware%20Programming%20Instructions.pdf) I have to install mksLaserTool softer on my computer.
I can't manage to find the software. Is there any link where I can find it or Does it exists another way to configure my board ?

RandomHacks-Git, 2021-12-10:
https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/tool

lgm42, 2021-12-11:
So simple ! Sorry for that ! Thanks

lgm42, 2021-12-11:
I have another question.
Folder you indicate contains the version 1.0.2 that havn't wifi configuration tool option.
How Can I configure the wifi ?

RandomHacks-Git, 2021-12-11:
I'm not sure about the tool, I'm not a collaborator of this repository.
You can configure wifi through the LCD, if you don't have one this might help: https://github.com/makerbase-mks/MKS-DLC32/issues/12

lgm42, 2021-12-11:
I finally find a solution.
With serial link I send commands:

> $Sta/SSID=wifi name
> $Sta/Password=wifi password
> $Radio/Mode=STA

closed 2021-12-11 by lgm42