makerbase-mks/MKS-DLC32 #73: GRBL PIN/FUNCTION MISSED

Cortapalos, 2021-12-11:
In grbl there are 2 functions that I miss on this board. One is the spindle direction control which can also be configured as Laser Enable, and another is the Coolant Enable gcode M8 which can also be used with air assist control. These pins come on the 8bits DLC V1.0 board, why not include them? In arduino it corresponds to pins D13 and A3. Also missing are the inputs for START, PAUSE / DOOR OPEN buttons (A1, A2)
I attach a photo of an 8-bit arduino and how GRBL was thought 
[image]
[image]

FrancescoIT, 2021-12-12:
and for some time we have been asking for a firmware change for this problem but still no results #53

makerbase-mks, 2021-12-13:
@Cortapalos , new released [V2.0.7_H35_20211213_N_CNC](https://github.com/makerbase-mks/MKS-DLC32/blob/main/firmware/TS35/CNC/Normal/Board_v2.0/V2.0.7_H35_20211213_N_CNC.bin) has added the M7/M8/M9 function(using SCL pin). the spindle direction has no IOs left, we should decide which IO should be used.

Cortapalos, 2021-12-13:
Great job thanks

FrancescoIT, 2021-12-13:
thanks great job, I run to update !!!

closed 2021-12-13 by Cortapalos