makerbase-mks/MKS-DLC32 #74: MKS-DLC32 updata

una2012, 2021-12-12:
I have been trying to update MKSDLC 2 for several days, but without success.
Can someone help me how and which FV to install.
This was read from the ICC
FW version: 1.1 (2021051202)
VER: 1.1.2021051202:]
[OPT: PHSW]
[MSG: Using machine: MKS DLC32]
[MSG: No Wifi]

makerbase-mks, 2021-12-13:
@una2012 Did you update according to the [method](https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware) ?Can you descripe what is the phyomenon?

Bobertsawesome, 2021-12-14:
I had the same issue, you must use an older version of ESP32 flash download tool, specifically, 3.6.6.

@makerbase-mks I suggested updating your github to provide this download
https://esp32.com/download/file.php?id=3335

Bobertsawesome, 2021-12-14:
V3.9.2 listed on https://www.espressif.com/en/support/download/other-tools is not working.

solawc, 2021-12-14:
The official flash_download_tool is better compatible with the old version. We have developed a tool that can be used for downloading @Bobertsawesome @una2012 https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/tool

una2012, 2021-12-14:
I tried both tools but without success. I'll try the older version, so I'll let you know what I did and whether I succeeded

**@Bobertsawesome** Thanks for the link for the old version, I'll let you know what happened

solawc, 2021-12-14:
If it still fails to download, please short-circuit SDA and GND, then power on again, and then use the tool to download @una2012

una2012, 2021-12-14:
I managed to do an update to the new version
`$I
[VER:1.1.2021121301:]
[OPT:MPHSW]
[MSG:Using machine:MKS DLC32]
[MSG:Mode=STA:SSID=Dejan:Status=Connected:IP=192.168.1.10:MAC=94-3C-C6-67-1D-2C]
ok`
https://imagizer.imageshack.com/v2/1024x768q90/922/bUgQBe.png

I managed to connect to the wifi from the computer

https://imagizer.imageshack.com/v2/1024x768q90/922/lKHZst.png
**But I still can't connect from my Android phone**
https://imagizer.imageshack.com/v2/1024x768q90/924/gsyaX8.jpg
https://imagizer.imageshack.com/v2/1024x768q90/923/F1Cg0E.jpg
https://imagizer.imageshack.com/v2/1024x768q90/924/rJPe44.jpg

makerbase-mks, 2021-12-15:
@una2012 Is your phone in the same LAN of the dlc32?

una2012, 2021-12-15:
> @una2012 Is your phone in the same LAN of the dlc32?

Yes, I have tried on several devices that are on the same LAN and will not connect to any of them.
All my phones and tablets are on the same IP from 192.168.1.1 to 192.168.1.255

electrontube, 2021-12-16:
I tried exactly the same as @una2012 
[image]

unfortunately with this result only:

[image]

Bobertsawesome, 2021-12-16:
Try putting a 10uf capacitor across EN and Ground pins. Refer to the ESP32 pin out to see which is which, but it should be pin 1 for ground and pin 3 for EN.

solawc, 2021-12-17:
I saw your picture, so I want to confirm some questions with you:
1. Do you use DLC32 V1.0 motherboard? Or use DLC32 V2.0 motherboard?
2. Is there a connection when downloading? For example, probe is connected? Or limit?
3. Has the motherboard been completely erased? @ @electrontube @una2012

una2012, 2021-12-17:
greeting i use dlc32 v2.0. pl9cica is not yet connected to motors and limit switches. only the power supply and usb cable are connected to the computer. entered data for my network. after that I turn off uwb from computer and google chrome I type 192.168.1.9 a laser control page appears in the browser. card how sqm got i just installed a new version of fw. 2.0.7

una2012, 2021-12-17:
Do I need to completely delete the dlc32 card and how do I do it?

electrontube, 2021-12-17:
> I saw your picture, so I want to confirm some questions with you:
> 
> 1. Do you use DLC32 V1.0 motherboard? Or use DLC32 V2.0 motherboard?

_It's DLC32 v2.1_

> 2. Is there a connection when downloading? For example, probe is connected? Or limit?

_There is nothing but the display connected to the board_

> 3. Has the motherboard been completely erased? @ @electrontube @una2012

_yes, (presuming the erase button on the MKS tool does what it says)_

electrontube, 2021-12-17:
> Try putting a 10uf capacitor across EN and Ground pins. Refer to the ESP32 pin out to see which is which, but it should be pin 1 for ground and pin 3 for EN.
> […](#)

_tried 10uF between GND and EN -> same result_

makerbase-mks, 2021-12-18:
@electrontube Have you tried this method: https://github.com/makerbase-mks/MKS-DLC32/issues/19#issuecomment-978785658  ？

electrontube, 2021-12-18:
I will give it a try, when I'm back to my desk.
I would assume, when the chip doesn't enter the flash mode, the upload would fail.
In my case the upload was succesful, but the fw crashes.

see screenshots above

electrontube, 2021-12-20:
I have found a workaround!
I have tried esptool.py on a linux machine and have sucessfully flashed the 2.07 firmware.
(seems Windows was causing the problems)

una2012, 2021-12-20:
Should it say VER: 2.0.7
Or as with me VER: 1.1.2021121301

> $I [VER:1.1.2021121301:] [OPT:MPHSW] [MSG:Using machine:MKS DLC32] [MSG:Mode=STA:SSID=Dejan:Status=Connected:IP=192.168.1.10:MAC=94-3C-C6-67-1D-2C] ok

Did I even manage to do the update to version 2.0.7?

una2012, 2021-12-20:
@electrontube 

> I have found a workaround! I have tried esptool.py on a linux machine and have sucessfully flashed the 2.07 firmware. (seems Windows was causing the problems)

Can you read $ I and ask to see which version is written in your code after you have successfully done the update

electrontube, 2021-12-20:
@una2012 

_$I 
[VER:1.1.2021121301:] 
[OPT:MPHSW] 
[MSG:Using machine:MKS DLC32] 
[MSG:Mode=AP:SSDI=MKS_DLC:IP=***...]_

una2012, 2021-12-22:
Greeting,

Again, with questions and problems.
I did an update to the new FW V2.0.8_H35_20211222_N.
I also did a mobile app update to version V 2.0.25
I have a problem again with connecting the mobile phone to the ICC DLC32.
I can't connect via the mobile app.
I successfully connect to the IP address 192.168.1.10 via a computer

I tried on more android phones and tablets but without success.

I would beg for help with connecting

$$ list

> $Sta/SSID=Dejan
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=MKS_DLC
$AP/Password=******
$AP/IP=192.168.4.1
$AP/Channel=1
$System/Hostname=grblesp
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=STA
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=XY
$Homing/Cycle1=
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=10
$Stepper/IdleTime=25
$Stepper/StepInvert=
$Stepper/DirInvert=X
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=Off
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=Off
$Homing/DirInvert=X
$Homing/Squared=
$Homing/Feed=300.000
$Homing/Seek=1000.000
$Homing/Debounce=250.000
$Homing/Pulloff=1.000
$Spindle/PWM/Frequency=1000.000
$GCode/MaxS=1000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=On
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=1000.000
$Spindle/Type=LASER
$flame_status=Off
$gyro_status=Off
$beep_status=Off
$Language=0
$rollerMode=On
$X/StepsPerMm=80.000
$Y/StepsPerMm=80.000
$Z/StepsPerMm=80.000
$A/StepsPerMm=100.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=6000.000
$Y/MaxRate=6000.000
$Z/MaxRate=6000.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=500.000
$Y/Acceleration=500.000
$Z/Acceleration=500.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=285.000
$Y/MaxTravel=272.000
$Z/MaxTravel=80.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=0.250
$Y/Current/Run=0.250
$Z/Current/Run=0.250
$A/Current/Run=0.250
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=0.125
$Y/Current/Hold=0.125
$Z/Current/Hold=0.125
$A/Current/Hold=0.125
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=16
$Y/Microsteps=16
$Z/Microsteps=16
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
Stream completed in 0:00

una2012, 2021-12-24:
Can someone who switched to ver 2.0.8 hang $$ fail
And to whom the web works via computer and mobile application mkslaser v 2.0.25

Cncninja, 2021-12-31:
I updated but had to use the MKS update tool, the ESP flash download tool did not work with the instructions provided. but the MKSlaser tool for updating worked like a charm. BUT.... you need to read the text in the instruction as they don't match the picture. I.E need to match the text "spi speed 80Mhz, Flash size 8MB, baud 115200" the pics so 40Mhz and 4mb and did not work the first time i tried i changed to the specs listed in the text and worked like a champ. I.E text is correct pics are wrong.

una2012, 2022-01-02:
I managed to connect mkslaser from the phone to MKSDLC32.
I went from scratch to delete the app on my phone, deleted the MKSDLC32 board and installed everything from scratch. Everything worked

closed 2022-07-04 by abinoer