makerbase-mks/MKS-DLC32 #78: Technical data of the ports of the MKS dlc32

Devil-67, 2021-12-17:
Can anyone tell me how big the current load is for this connection? 
An 8A power supply unit is connected to my controller.
I would like to supply my diode laser Neje a40640 with power at this connection.
What do you think about supplying the laser directly with voltage via the controller?

[image]

Plottmania, 2021-12-18:
Power supply with 12 volts and 12 volts come out of there.
Power supply with 24 volts and 24 volts come out of there...so be careful!
Currently there is a Sculpfun S9 and the imprint 24 volts but whether this really all works so great if the fan on the laser only tolerates 12 volts...?

Devil-67, 2021-12-18:
My power supply has 12V and 8 amps.
The 12V supply voltage is also connected to the port.
My question is how much current can I draw from this port?

makerbase-mks, 2021-12-18:
@Devil-67 Max current is about 3A, please be sure use under this current.

Devil-67, 2021-12-18:
Then it is probably better to supply the Neje laser separately.
The Neje is specified with 3-4 Amps.

closed 2021-12-27 by makerbase-mks