makerbase-mks/MKS-DLC32 #79: SD Card causes board to Reset

pnt-thiago, 2021-12-18:
Hi, I'm experiencing some issues when I have an SD card connected the board reboot at random intervals.
Firmware is latest: V2.0.7_H35_20211213

torenich, 2021-12-19:
I am having this same issue with two separate boards. Both DLC32 V2.1_002. One with a TS24-R V2.0 and one with a TS35-R V2.0. Both with latest firmware. I have tried multiple SD cards, 128MB, 1GB, and 2GB.

solawc, 2021-12-20:
Can I ask you some questions?
1. Is the SD card you use Class4 or Class10?
2. What is the capacity of the SD card?
3. Does this happen when the card is not inserted?

torenich, 2021-12-20:
Thank you for your response. I do not know what class they are#1 is 128MB, #2 is 1GB, #3 is 2GB. #1 un-branded, came with Monoprice 3D printer, #2 and #3 are Sandisk. See attached picture.No. But it only starts to happen when the SD card is accessed. If I try to upload a gcode file, is fails, then starts rebooting. Tim Sent from Mail for Windows From: SolaSent: Sunday, December 19, 2021 9:02 PMTo: makerbase-mks/MKS-DLC32Cc: torenich; CommentSubject: Re: [makerbase-mks/MKS-DLC32] SD Card causes board to Reset (Issue #79) Can I ask you some questions?Is the SD card you use Class4 or Class10?What is the capacity of the SD card?Does this happen when the card is not inserted?

pnt-thiago, 2021-12-20:
> Can I ask you some questions?
> 
> 1. Is the SD card you use Class4 or Class10?
> 2. What is the capacity of the SD card?
> 3. Does this happen when the card is not inserted?

Hi Solawc. I teted with a Class 2 with 129Mb and a Class 10 with 16Gb, both of them caused the reset.
No, when I am not using any sd card, the board does not reset.

Ondrashek, 2021-12-20:
I have this issue during engraving. When I turn offline engraver again it goes well.

@makerbase-mks, @solawc  Is there any debug file, what's happened?

torenich, 2021-12-23:
Just flashed firmware 2.0.8 to both boards and SD card access now working correctly.

pnt-thiago, 2021-12-23:
I confirm that latest firmware solved this issue. Thanks for the assistance!

closed 2021-12-23 by pnt-thiago

keithstrachan, 2022-07-15:
looks like this issue is back, I am having it on v2.0.8_h35 & v2.10_h35 series