makerbase-mks/MKS-DLC32 #80: Change LCD display GUI 5% laser is too high, need 0.2%

KenDodman, 2021-12-19:


solawc, 2021-12-20:
Because when we tested, we used a 3W/5W laser spindle for testing, its output was not very amazing; but maybe we can add configuration items in the next version to make the power of low light/high light configurable

Watto2021, 2022-01-16:
I would like the same user configure the Low power %. 
The new 10W diode laser available now burns holes in cardboard and thin wood at the default power

forest-live, 2022-02-16:
The problem with the new firmware has not disappeared! How to reduce the laser power in 5% mode?

forest-live, 2022-02-18:
Are these official settings or just your assumptions?

hallskelet, 2022-05-26:
Is it possible to make this setting changeable through the web interface or config file?
Write "low power" on the button, and which one to configure in the program.
And is it possible to make it so that this mode does not turn off by itself? this is not described in the documentation and is very inconvenient when you need to set up a solid-state laser, although it is quite understandable if you work with a CO2 laser.