makerbase-mks/MKS-DLC32 #82: documentation on DLC 32 V2.0

killrob, 2021-12-19:
Can we please know the function of dip switches in the V2.0?
i have found no infos on that.

solawc, 2021-12-20:
> 

Yes, the detailed description is in production, and you will get it soon

Plottmania, 2021-12-20:
This is quite simple : )
Only with Pololu 4988 and 8825 motor drivers the dips work.
So the step resolution can be set here.
With TMC drivers it does not matter what is jumpered there like...

killrob, 2021-12-20:
> This is quite simple : ) Only with Pololu 4988 and 8825 motor drivers the dips work. So the step resolution can be set here. With TMC drivers it does not matter what is jumpered there like...

And it's ok i have 4899 drivers, the  question is the meaning of the dips, not with which drivers they works

Plottmania, 2021-12-20:
I hope it helps you a little...

https://plottmania.de/forum/thread/123-mks-dlc-stepper-resolution/

[image]

Plottmania, 2021-12-22:
...
[image]

makerbase-mks, 2022-01-07:
Here is the document: https://github.com/makerbase-mks/MKS-DLC32/blob/main/doc/DLC32%20wiring%20manual.pdf

closed 2022-01-07 by makerbase-mks