makerbase-mks/MKS-DLC32 #87: Firmware for the dlc32 v 2.1

Earloffunk, 2021-12-21:
Hello i Buy a dlc32 v2.1 for my Laser Graver with a ts35 Tft. And i find no Firmware. Can i use the Firmware from the 2.0 Version?

Ondrashek, 2021-12-22:
Just read a readme:

Version description

E.g：

V2.0.3_H35_20211108_N

V2.0.3: Firmware version

H35: Horizontal screen- 3.5Inch

20211108： Modified date （2021-11-08）

N：N for normal machine, C for CoreXY machine

(for DLC V2.x, coreXY): As there is a little difference between DLC V1.x and V2.x boards, so we would explain what board the firmware is suitable for. Please use the correct version, otherwise it may not work.

manuelchaves, 2021-12-22:
Hi, its ok to V2.1 hardware version, its for 2,X (2.0, 2.1 and etc), ok?

Ondrashek, 2021-12-22:
> Hi, its ok to V2.1 hardware version, its for 2,X (2.0, 2.1 and etc), ok?

Correct

closed 2021-12-27 by makerbase-mks

GordonS242, 2023-02-12:
I have a DLC32 v2.1 I find that fulidnc and grblhal. Will not work on this card however the software you provide will work on the card can you please tell me why fulidnc will not work

nofuturekid, 2023-02-12:
Both, FluidNC and GRBLhal are both working fine with a DLC32 v2.0/2.1
Only the LCD is not working but everything else works beautifully.

GordonS242, 2023-02-12:
I have just downloaded the latest fluidnc installed and it still does not work however i used
MKS_DLC_21_XYZ.YAMI File ust to see if it would work unfortunately it does not work
any suggestion
Regards,
Gordon.stuart@optusnet.com

nofuturekid, 2023-02-12:
What did not work? I had it installed, everything worked but I was missing my display too much on the long run so I reverted back to GRBL.

GordonS242, 2023-02-12:
All fixed you you made me re-think and I found the problem it now works thank you.

philvtx, 2023-02-15:
> Tout t'a réparé tu m'as fait repenser et j'ai trouvé le problème ça marche maintenant merci.

bonjour j'ai un problème avec ma carte mks dlc32 sur un laser diode je n'ai qu'1 référence sur les 3 ! j'ai  X ou Y le plus prés des switch au moment de la référence mais le Z ne se fait pas quand je valide hard switch le moteur Z se coupe au contact du switch en déplacement manuel