makerbase-mks/MKS-DLC32 #88: firmware V2.0.8_H35_20211222_N_CNC.bin not home Z

ClaudeHerzig, 2021-12-24:
the firmware V2.0.8_H35_20211222_N_CNC.bin
the $H no longer works for Z

makerbase-mks, 2022-01-07:
Will be fixed recently.

ClaudeHerzig, 2022-01-07:
firmware V2.0.8_H35_20211222_N_CNC.bin and V2.0.8_H35_20220105_N_CNC.bin

the $ H no longer works for Z

On firmware V2.0.7_H35_20211213_N_CNC.bin the $ H works well for X, Y, Z

toonaoki, 2022-01-07:
Firmware : V2.0.8_H35_20220105_N_CNC.bin
I also checked with the new firmware, but it seems that Z-axis homing is disabled.

toonaoki, 2022-01-08:
I found a solution.
There are setting items that determine the order of Homing.

defaults: defaults:
> $ Homing
$ Homing / Cycle0 = XY
$ Homing / Cycle1 =
$ Homing / Cycle2 =
$ Homing / Cycle3 =
$ Homing / Cycle4 =
$ Homing / Cycle5 =
$ Homing / Enable = On
$ Homing / DirInvert =
$ Homing / Squared =
$ Homing / Feed = 300.000
$ Homing / Seek = 1000.000
$ Homing / Debounce = 250.000
$ Homing / Pulloff = 1.000

After editing:
$ Homing / Cycle0 = Z
$ Homing / Cycle1 = XY

With this change, homing works with the latest version of firmware.

ClaudeHerzig, 2022-01-08:
Thanks its function well with the latest firmware

$Homing/Cycle0=Z
$Homing/Cycle1=X
$Homing/Cycle2=Y

abinoer, 2022-01-15:
The latest firmware can make $H work on the Z axis by sending instructions, please update
Send "$Homing/Cycle1=Z" on, send "$Homing/Cycle1=0" off

closed 2022-01-15 by abinoer