makerbase-mks/MKS-DLC32 #91: seite geht nicht

compi9er, 2021-12-31:
hallo die seiten

https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/Laser/Normal

https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/Laser/CoreXY

https://github.com/makerbase-mks/MKS-DLC32/tree/main/firmware/CNC/Normal

gehen nicht, kommt immer was mit fehler 404

mfg m. neuner

trashypopashy, 2022-01-01:
Klick dich einfach durch den Verzeichnisbaum. Die haben ihre Struktur verändert, es ist jetzt deutlich besser sortiert. Du musst im Ordner Firmware dein Display wählen (2,4 oder 3,5 Zoll), dann wählst du Laser oder eben CNC, dann deine Board-Revision (1 oder 2) und dann wirst du fündig.

compi9er, 2022-01-01:
hallo
es geht um das board und nicht um das display
mfg m. neuner

Verschickt: Sa, 1. Jan. 2022 12:01

Plottmania, 2022-01-01:
Du hast aber schon gelesen wie das funktioniert?
Also ich spiel da egal was drauf, es funktioniert.
Je nach dem welches Display die Version passend zum Display.
Kein Display vorhanden kannst Du selber wählen nur sollte man drauf achten zwischen Board-Revision 1 und 2 gibt es kleine Unterschiede...kann man auch erlesen : )
Also wählen, flashen und spielen...ganz einfach.