makerbase-mks/MKS-DLC32 #94: How to setup microsteps

killrob, 2022-01-04:
Hi all, i want to know how setup jumpers on a DLC32 V1.1 with TMC2225 drivers to obtain 1/32 microstep, i've found no docs on it.
Thanks

Ondrashek, 2022-01-04:
Hi,

You did not find deeper. 

https://github.com/makerbase-mks/MKS-DLC32/blob/main/doc/DLC32%20wiring%20manual.pdf

killrob, 2022-01-04:
> Hi,
> 
> You should search deeper.
> 
> https://github.com/makerbase-mks/MKS-DLC32/blob/main/doc/DLC32%20wiring%20manual.pdf

no, i think YOU did not undertsand. You linked me the manual on the DLC32 V2.1, i asked for the V1.1 that HAS NOT the dip switch but jumpers.

Ondrashek, 2022-01-04:
> > Hi,
> > You should search deeper.
> > https://github.com/makerbase-mks/MKS-DLC32/blob/main/doc/DLC32%20wiring%20manual.pdf
> 
> no, i think YOU did not undertsand. You linked me the manual on the DLC32 V2.1, i asked for the V1.1 that HAS NOT the dip switch but jumpers.

Just think. Jumper - HIGH, no jumper LOW. You want 1/32 microsteps on TMC2225 so M0, M1 to HIGH HIGH. You can set M2 to high as well.

makerbase-mks, 2022-01-07:
> > > Hi,
> > > You should search deeper.
> > > https://github.com/makerbase-mks/MKS-DLC32/blob/main/doc/DLC32%20wiring%20manual.pdf
> > 
> > 
> > no, i think YOU did not undertsand. You linked me the manual on the DLC32 V2.1, i asked for the V1.1 that HAS NOT the dip switch but jumpers.
> 
> Just think. Jumper - HIGH, no jumper - LOW. DIP switch is the same ON is jumpered. You want 1/32 microsteps on TMC2225 so M0, M1 to HIGH HIGH. You can set M2 to high as well but it will not be used.

@killrob Yes, just as what @Ondrashek said.

closed 2022-01-07 by makerbase-mks