makerbase-mks/MKS-DLC32 #95: Lcd

mikerocklewitz, 2022-01-04:
If i wanted to use my own resistive lcd panel, what kind and how is it wired?

makerbase-mks, 2022-01-07:
Make sure your lcd panel supports SPI interface and refer to the pins here:https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V2.1_003/MKS%20DLC32%20V2.1_003%20SCH.pdf

closed 2022-01-07 by makerbase-mks

mikerocklewitz, 2022-09-17:
> Make sure your lcd panel supports SPI interface and refer to the pins here:https://github.com/makerbase-mks/MKS-DLC32/blob/main/hardware/MKS%20DLC32%20V2.1_003/MKS%20DLC32%20V2.1_003%20SCH.pdf

I need to reopen the question. 8 months after asking im finally modifying a 3018.....

What type of lcd is being used? I see the pinout for spi in the schematics but what is the lcd controller so i use the correct kind.