makerbase-mks/MKS-DLC32 #97: Z axis limit switch not working? (2.4" lcd, laser version)

RandomHacks-Git, 2022-01-06:
Hi, I'm trying to use a limit switch for the Z axis but the board doesn't seem to react to it, no alarm is triggered.
I tried on the "Z" connector (IO34) and the "Probe" connector (IO22) without success. 

I'm using an NO switch connected between GND and the IO pin (I'm using the same configuration for the X and Y axes which work fine). I also tried changing $6 but it didn't make any difference.

I have v2.0 of the board with the 2.4" LCD and laser firmware.

Help?

MKS-hosiey, 2022-01-07:
We will update the firmware soon, and the new firmware will fix the problem

solawc, 2022-01-12:
The latest firmware can be used normally, please update

RandomHacks-Git, 2022-01-12:
Works fine now, thank you.

closed 2022-01-12 by RandomHacks-Git